realsenseai/realsense-ros #372: Depth not configurable in D435

capstayn, 2018-04-21:
I'm using realsense D435. On ubuntu 16.04.1, kernel 4.13.0-38, x86_64, ROS kinetic, librealsense 2.10.4 (installed from source), realsense-2.0.3 installed in catkin_ws/src. 

When executing
```
roslaunch realsense2_camera rs_camera.launch
```
The dimensions and frequency of depth, published on the topic `/camera/depth/image_rect_raw`, cannot be changed by the configuration set in `rs_camera.launch`. If `depth_width=640`, `depth_height=480`, the `fps` published in the topic will always be 30 fps regardless its value in `rs_camera.launch`. If the dimensions are different from `depth_width=640`, `depth_height=480`, the topic `/camera/depth/image_rect_raw` will not publish any message at all.

I can change without problem the resolution and frequency of `depth` in the SDK of Intel, so it seems that the library is well installed and functioning. When I launch `rs_camera.launch` ROS messages notify that the desired resolution of depth has been set up, and there is no error thrown (apart from `WARNING [139934683240320] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error` which is always printed out anyway).

capstayn, 2018-04-22:
I figured out that  infrared images, color image and depth image should be all set with the same dimensions in order for the desired resolution to be published. I guess I should have known this... Perhaps `realsense_camera2_ros` should enforce this pattern?
In any case, I also noticed that all the topics are published with the same `fps`, and if they are set to different frequencies, the lowest `fps` is actually set. 
Also, on the SDK by Intel it's possible for `color image` to be in a different resolution than `depth image`, specifically the full HD resolution (1920x1080). I haven't managed to get this configuration working (`color_image` at 1920x1080 and `depth_image` at 1280x720). 

Would it be possible to decouple the resolution and fps of `color image` and `depth image`?

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Close this issue for lack of activity. Please feel free to reopen if the issue is still relevant.

closed 2019-03-27 by RealSenseCustomerSupport

zzy729425207, 2021-05-31:
Hello,I want to change the "color_width" and "color_height" parameter of D435. But whatever the parameter I changed in rs_camera.launch, the output resolution always be 640*480.