realsenseai/realsense-ros #373: Jetson TX2

sallvadorrr, 2018-04-23:
Hello,

I am trying to run: _catkin_make clean_ on Jetson TX2, after installing: https://github.com/jetsonhacks/installRealSenseROSTX2

but I get the following error:

nvidia@tegra-ubuntu:~/catkin_ws$ catkin_make clean
Base path: /home/nvidia/catkin_ws
Source space: /home/nvidia/catkin_ws/src
The specified base path "/home/nvidia/catkin_ws" contains a CMakeLists.txt but "catkin_make" must be invoked in the root of workspace
nvidia@tegra-ubuntu:~/catkin_ws$ catkin_make clean
Base path: /home/nvidia/catkin_ws
Source space: /home/nvidia/catkin_ws/src
Build space: /home/nvidia/catkin_ws/build
Devel space: /home/nvidia/catkin_ws/devel
Install space: /home/nvidia/catkin_ws/install
####
#### Running command: "cmake /home/nvidia/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/nvidia/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/nvidia/catkin_ws/install -G Unix Makefiles" in "/home/nvidia/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/nvidia/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nvidia/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.11
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - realsense2_camera
-- ~~  - realsense_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "diagnostic_updater" with any of the following names:

    diagnostic_updaterConfig.cmake
    diagnostic_updater-config.cmake

  Add the installation prefix of "diagnostic_updater" to CMAKE_PREFIX_PATH or
  set "diagnostic_updater_DIR" to a directory containing one of the above
  files.  If "diagnostic_updater" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  realsense2_camera/CMakeLists.txt:6 (find_package)

-- Could not find the required component 'diagnostic_updater'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "diagnostic_updater" with any of the following names:

    diagnostic_updaterConfig.cmake
    diagnostic_updater-config.cmake

  Add the installation prefix of "diagnostic_updater" to CMAKE_PREFIX_PATH or
  set "diagnostic_updater_DIR" to a directory containing one of the above
  files.  If "diagnostic_updater" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  realsense2_camera/CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Could you please help me with that issue?

icarpis, 2018-04-27:
Hello @sallvadorrr, the `diagnostic updater` package should be a part of the ROS Kinetic installation.
Anyway, please try to install it manually by `sudo apt-get install ros-kinetic-diagnostic-updater`

closed 2018-06-07 by sallvadorrr

rakscc, 2022-08-10:
> Hello @sallvadorrr, the `diagnostic updater` package should be a part of the ROS Kinetic installation. Anyway, please try to install it manually by `sudo apt-get install ros-kinetic-diagnostic-updater`

I got the same error for ROS2 Humble version. It was fixed with the command `sudo apt-get install ros-humble-diagnostic-updater` Thank you.

monajalal, 2023-10-24:
@rakscc thank you
```
(base) mona@mona-ThinkStation-P7:~/realsense-ros$ colcon build
Starting >>> realsense2_camera_msgs
Finished <<< realsense2_camera_msgs [2.00s]                     
Starting >>> realsense2_camera
Starting >>> realsense2_description
Finished <<< realsense2_description [1.03s]                                                               
--- stderr: realsense2_camera                         
CMake Error at CMakeLists.txt:117 (find_package):
  By not providing "Finddiagnostic_updater.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "diagnostic_updater", but CMake did not find one.

  Could not find a package configuration file provided by
  "diagnostic_updater" with any of the following names:

    diagnostic_updaterConfig.cmake
    diagnostic_updater-config.cmake

  Add the installation prefix of "diagnostic_updater" to CMAKE_PREFIX_PATH or
  set "diagnostic_updater_DIR" to a directory containing one of the above
  files.  If "diagnostic_updater" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< realsense2_camera [2.07s, exited with code 1]

Summary: 2 packages finished [4.71s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera
(base) mona@mona-ThinkStation-P7:~/realsense-ros$ sudo apt-get install ros-humble-diagnostic-updater
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  cuda-cccl-11-7 cuda-command-line-tools-11-7 cuda-compiler-11-7 cuda-cudart-11-7 cuda-cudart-dev-11-7 cuda-cuobjdump-11-7 cuda-cupti-11-7 cuda-cupti-dev-11-7 cuda-cuxxfilt-11-7 cuda-demo-suite-11-7 cuda-documentation-11-7 cuda-driver-dev-11-7 cuda-gdb-11-7 cuda-libraries-11-7
  cuda-libraries-dev-11-7 cuda-memcheck-11-7 cuda-nsight-11-7 cuda-nsight-compute-11-7 cuda-nsight-systems-11-7 cuda-nvcc-11-7 cuda-nvdisasm-11-7 cuda-nvml-dev-11-7 cuda-nvprof-11-7 cuda-nvprune-11-7 cuda-nvrtc-11-7 cuda-nvrtc-dev-11-7 cuda-nvtx-11-7 cuda-nvvp-11-7 cuda-runtime-11-7
  cuda-sanitizer-11-7 cuda-toolkit-11-7 cuda-toolkit-11-7-config-common cuda-tools-11-7 cuda-visual-tools-11-7 gds-tools-11-7 libcublas-11-7 libcublas-dev-11-7 libcufft-11-7 libcufft-dev-11-7 libcufile-11-7 libcufile-dev-11-7 libcurand-11-7 libcurand-dev-11-7 libcusolver-11-7
  libcusolver-dev-11-7 libcusparse-11-7 libcusparse-dev-11-7 libnpp-11-7 libnpp-dev-11-7 libnvidia-egl-wayland1 libnvjpeg-11-7 libnvjpeg-dev-11-7 nsight-compute-2022.2.1 nsight-systems-2022.1.3
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  ros-humble-diagnostic-updater
0 upgraded, 1 newly installed, 0 to remove and 4 not upgraded.
Need to get 98.9 kB of archives.
After this operation, 391 kB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-diagnostic-updater amd64 3.1.2-1jammy.20230919.223534 [98.9 kB]
Fetched 98.9 kB in 1s (173 kB/s)                         
Selecting previously unselected package ros-humble-diagnostic-updater.
(Reading database ... 397403 files and directories currently installed.)
Preparing to unpack .../ros-humble-diagnostic-updater_3.1.2-1jammy.20230919.223534_amd64.deb ...
Unpacking ros-humble-diagnostic-updater (3.1.2-1jammy.20230919.223534) ...
Setting up ros-humble-diagnostic-updater (3.1.2-1jammy.20230919.223534) ...
(base) mona@mona-ThinkStation-P7:~/realsense-ros$ colcon build
Starting >>> realsense2_camera_msgs
Finished <<< realsense2_camera_msgs [1.17s]                     
Starting >>> realsense2_camera
Starting >>> realsense2_description             
Finished <<< realsense2_description [0.27s]                                                               
Finished <<< realsense2_camera [16.6s]                       

Summary: 3 packages finished [18.4s]
```