realsenseai/realsense-ros #375: How to toggle Depth Emitter from another node?

GuidoBartoli, 2018-04-26:
I'm using a D435 with the `realsense2_camera` ROS package (v.2.0.3).
I saw that dynamic reconfigure can access many camera parameters as well as enabling/disabling the IR Depth Emitter.
Is there a way to toggle it even from another running node (or launch file)? I could not find any external parameter for `realsense2_camera` node to be changed in real time...

Thanks

comkieffer, 2018-04-27:
You're lucky, I did that just yesterday. 

```
#include <dynamic_reconfigure/client.h>

// Global variable for the example
using realsense_dyn_reconfigure = dynamic_reconfigure::Client<realsense2_camera::rs435_paramsConfig>;
boost::shared_ptr<realsense_dyn_reconfigure> realsense_params;

bool enable_realsense_emitter(bool enable) {
    // Off (disabled) -> 0, On (enabled) -> 1
    int parameter_value = (enable)? 1 : 0;
    
    realsense2_camera::rs435_paramsConfig cfg;
    cfg.rs435_depth_emitter_enabled = parameter_value;
    if (!realsense_params->setConfiguration(cfg)) {
        ROS_ERROR("Unable to set realsense camera configuration");
        return false;
    }
    
    return true;
}

int main( ... ) {
    ...
    realsense_params = boost::make_shared<realsense_dyn_reconfigure>("/realsense/realsense2_camera_manager/");
    ...
    enable_realsense_emitter(false);
    ....

    return 0;
}

```

You also need the `realsense2_camera/rs435_paramsConfig.h` from the `devel` folder file which unfortunately isn't exported by the realsense2_camera package. I have made a PR to fix this (#377) but in the mean time you can copy it over to your own package. 

You may also need to copy the `rs435_paramsConfig.h` file generated by the package to your own package and include it. I have just made a PR to update the `CMakeLists.txt` to solve this however.

stwirth, 2018-04-27:
See http://wiki.ros.org/dynamic_reconfigure. There's a python API, a C++ API, a GUI (rqt_reconfigure) and a command line API (`rosrun dynamic_reconfigure dynparam set ...`). You can use the latter in launch files.

GuidoBartoli, 2018-05-10:
Thank you very much to you both! :)

closed 2018-05-10 by GuidoBartoli

GuidoBartoli, 2018-05-15:
Thank you very much @comkieffer, I managed to do it following your suggestion and [this](http://ros-robotics.blogspot.it/2016/03/controling-dynamic-reconfiguration-from.html) useful article!