realsenseai/realsense-ros #378: depth (post processing) in ROS?

Kyungdon, 2018-04-30:
Is it possible to activate the depth post-processing in ROS?

yuyu2172, 2018-05-11:
I was wondering the same thing.

janisa9, 2018-07-06:
I am also wondering the same thing, is it worth even to look for post processing options in this driver?

johannesburg, 2018-07-07:
I've added three of the four available post processing filters to the realsense ROS library (temporal, spatial, and hole filling filter) and am currently working on adding the decimation filter. If you'd like to use the code, feel free to use my [forked repo](https://github.com/johannesburg/uw-realsense). All parameters are configurable via the launch files.

ManMan88, 2018-12-18:
I think this [forked repo](https://github.com/johannesburg/uw-realsense) won't build with the newest librealsense2.

I get the following error when trying to make:

```
 /usr/include/librealsense2/hpp/rs_sensor.hpp:109:13: error: no match for call to ‘(rs2::asynchronous_syncer) (rs2::frame)’
              on_frame_function(frame{ fref });
```

doronhi, 2018-12-19:
Yeah, well, it’s really not up to date.

closed 2018-12-19 by doronhi