realsenseai/realsense-ros #381: D435 Dropping all frames when launching the nodelet

jacobmoroni, 2018-05-10:
I am using the D435. When I launch rs_rgbd.launch the first time, it works. After killing it once, however, it seems to drop all of the frames and wont work again unless I unplug the camera and plug it back in. I just updated the firmware to 5.9.9.2 and I have this problem on every computer that I run it on. 

I am killing it with ctrl+c. I don't know if that will make a difference. 

this is the error that I get when I launch the launch file

[ INFO] [1525993622.031882067] [/camera/realsense2_camera_manager]: publishStaticTransforms...
[ INFO] [1525993622.032091362] [/camera/realsense2_camera_manager]: RealSense Node Is Up!
 10/05 17:07:02,035 WARNING [139926939100928] (backend-v4l2.cpp:808) Empty frame has arrived.
 10/05 17:07:02,035 WARNING [139926939100928] (backend-v4l2.cpp:808) Empty frame has arrived.
 10/05 17:07:02,035 WARNING [139926939100928] (backend-v4l2.cpp:808) Empty frame has arrived.
 10/05 17:07:02,035 WARNING [139926939100928] (backend-v4l2.cpp:808) Empty frame has arrived.
 10/05 17:07:07,029 WARNING [139927029090048] (backend-v4l2.cpp:825) Frames didn't arrived within 5 seconds
 10/05 17:07:07,035 WARNING [139926947493632] (backend-v4l2.cpp:825) Frames didn't arrived within 5 seconds

Is there a way to fix that so that I don't have to unplug it every time?

josephduchesne, 2018-05-12:
This branch resets the device on startup: https://github.com/icarpis/realsense/tree/reset_dev

I've found sometimes it takes a few tries, so I wrote a python program that watches the topic output and gives the nodelet launchfile 10 seconds to output data or it resets it.

jacobmoroni, 2018-05-14:
I just downloaded this and saw that it is for the old realsense package not the new one (the d400 series) and I couldn't get it to work for the new one. Do you have this working for one of the D400 cameras?

jacobmoroni, 2018-05-14:
I figured it out. I just downloaded the realsense_node_factory.h and realsense_node_factory.cpp files from the include and src folders in the repository and replaced the ones in the new package. I also needed to change the references to realsense_ros_camera to realsense2_camera in both files. 

After I did that, I rebuilt the package and now the camera restarts on launch and nothing breaks, so hopefully this will continue to work in future releases.

closed 2018-05-14 by jacobmoroni