realsenseai/realsense-ros #386: Release to Melodic/Ubuntu 18.04?

doisyg, 2018-05-28:
Now that Melodic is out, is there any plan to release to Melodic/Ubuntu18.04 for for both previous gen Realsense and D400 series?
Any tentative schedule?

doisyg, 2018-08-28:
@doronhi Any update?

Tuebel, 2018-09-11:
I haven't read any official comments on this yet but I am actually running the ROS realsense camera on Melodic/Ubuntu 18.04 without any problems.
Make sure that you have the current librealsense installed as they supported Ubutunu 18.04 quite recently.

doisyg, 2018-09-19:
Hi @Tuebel 
Thanks for your answer, can you provide more details ? Which did you tested?
How did you install librealsense on ubuntu 18.04? You compiled it from source? (on 16.04 the package ros-kinetic-librealsense was enough I believe)

Tuebel, 2018-09-19:
Hi @doisyg 
I followed the instructions from https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md to install the librealsense Debian package.
After installing these packages I was able to compile the ros package from source as described in https://github.com/intel-ros/realsense#step-3-install-intel-realsense-ros-from-sources .

doisyg, 2018-09-20:
Reporting progress for usage of a R200 camera in ROS with Ubuntu 18.04 and Melodic (and hence librealsense and realsense_camera **v1**)

#Installation of librealsense: OK
`sudo apt-get install libglfw3-dev`
download and unzip https://github.com/IntelRealSense/librealsense/archive/v1.12.1.zip
add `#include <functional> `to librealsense-1.12.1/src/type.h (in order to compile with ubuntu 18.04 builtin gcc 7)
`cd librealsense-1.12.1`
`mkdir build`
`cd build`
`cmake ..`
`make && sudo make install`

doisyg, 2018-09-20:
Currently struggling with the compilation of the ros realsense_camera library (https://github.com/intel-ros/realsense/archive/indigo-devel.zip)
Probably linking issue with the system installed librealsense

doisyg, 2018-09-20:
Got it working by modifying CMakelists.txt:

#Ros realsense_camera
download https://github.com/intel-ros/realsense/archive/indigo-devel.zip
Modify CMakelists.txt as below and build
[CMakeLists.txt](https://github.com/intel-ros/realsense/files/2401961/CMakeLists.txt)

I added:
SET(USE_SYSTEM_LIBREALSENSE ON)

Replaced:  https://github.com/intel-ros/realsense/blob/ebc06d4be1800da82595a7ccb163f3b5b6e69b23/realsense_camera/CMakeLists.txt#L34 
by:

    FIND_PATH(realsense_INCLUDE_DIR librealsense HINTS "/usr/local/include")
    FIND_LIBRARY(realsense_LIBRARY realsense HINTS "/usr/local/lib")

    FIND_PATH(realsense_INCLUDE_DIRS librealsense HINTS "/usr/local/include")
    FIND_LIBRARY(realsense_LIBRARIES realsense HINTS "/usr/local/lib")

Commented the following lines (it was causing errors and I don't need it):
https://github.com/intel-ros/realsense/blob/ebc06d4be1800da82595a7ccb163f3b5b6e69b23/realsense_camera/CMakeLists.txt#L114-L118

And modified https://github.com/intel-ros/realsense/blob/ebc06d4be1800da82595a7ccb163f3b5b6e69b23/realsense_camera/CMakeLists.txt#L147
in: 
`install(TARGETS ${PROJECT_NAME}_nodelet `

A bit dirty but it works

rbeltranb, 2018-12-11:
Due to the fact that _sudo_ clears the environment variables setup by setup.bash, for me was necessary run _sudo visudo_ and add there the following line in order to run _make_ with _sudo_:

     Defaults        env_keep += "PYTHONPATH"

prabinrath, 2019-01-05:
failed during initialization: gstv4l2object.c(3722): gst_v4l2_object_set_format_full (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin18/GstV4l2Src:v4l2src1:
Call to S_FMT failed for GREY @ 640x481: Input/output error

This is the out put from cheese

hcostelha, 2019-01-16:
> Got it working by modifying CMakelists.txt:
> 
> #Ros realsense_camera
> download https://github.com/intel-ros/realsense/archive/indigo-devel.zip
> Modify CMakelists.txt as below and build
> [CMakeLists.txt](https://github.com/intel-ros/realsense/files/2401961/CMakeLists.txt)
> 
> I added:
> SET(USE_SYSTEM_LIBREALSENSE ON)
> 
> Replaced:
> 
> [realsense/realsense_camera/CMakeLists.txt](https://github.com/intel-ros/realsense/blob/ebc06d4be1800da82595a7ccb163f3b5b6e69b23/realsense_camera/CMakeLists.txt#L34)
> 
> Line 34 in [ebc06d4](/intel-ros/realsense/commit/ebc06d4be1800da82595a7ccb163f3b5b6e69b23)
> 
>  find_package(realsense REQUIRED) 
> 
> by:
> ```
> FIND_PATH(realsense_INCLUDE_DIR librealsense HINTS "/usr/local/include")
> FIND_LIBRARY(realsense_LIBRARY realsense HINTS "/usr/local/lib")
> 
> FIND_PATH(realsense_INCLUDE_DIRS librealsense HINTS "/usr/local/include")
> FIND_LIBRARY(realsense_LIBRARIES realsense HINTS "/usr/local/lib")
> ```

Use only 
```
FIND_PATH(realsense_INCLUDE_DIRS librealsense HINTS "/usr/local/include")
FIND_LIBRARY(realsense_LIBRARIES realsense HINTS "/usr/local/lib")
```
you do not need the other.

doisyg, 2019-02-25:
It is working for me with kernel 4.15 (with the modified CMakeLists posted before)
I tried to update to the 18.04.2 (kernel 4.18) and it is NOT working. Each R200 is recognized as 6 video* devices instead of 3 before. I am not going to investigate further and stay on kernel 4.15 for the moment. But I will happily follow any solutions to make it work with 4.18.

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

peci1, 2019-03-20:
Hey, releasing this to melodic (together with "bundling" librealsense2 into the ROS APT repos) would really help. Is it on track for you?

doronhi, 2019-03-20:
Releasing realsense2_camera for Melodic is planned but as it seems to work for many people as it is, so far it never got the full attention of being supported officially.

peci1, 2019-03-20:
>  work for many people as it is

Well, it's a difference if you just run `rosdep install` on your workspace and forget about all the complexities, or if you have to manually clone repos, build them etc...

doronhi, 2019-03-20:
I absolutely agree and hope to manage that packaging soon.

SteveMacenski, 2019-03-20:
Thanks! I know this is a blocker on my end for Melodic transitions

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

HI SteveMacenski,

We understand the importance of the issue to you.
So far we have no further update yet. You may check back later on.

Thanks!

timrae, 2019-04-24:
Please release

doisyg, 2019-05-07:
I made the legacy librealsense work with 4.16+ kernel (and hence ubuntu 18.04.2) in this PR: https://github.com/IntelRealSense/librealsense/pull/3929
Hosted here: https://github.com/doisyg/librealsense/tree/recent_kernel_compatibility
Tested only with a R200, feel free to PR for other cams if needed but I won't be able to check that it works.

marcmateumateus, 2019-05-08:
Tested @doisyg's solution on SR300 with 4.18 kernel and it works.

peci1, 2019-06-05:
Why was this issue closed? I don't see any melodic release in rosdistro...

atyshka, 2019-07-30:
Yeah how difficult would it be to get this added to the buildfarm? Having to rebuild every time a new librealsense is tedious and having a melodic debian would be much easier.

fmessmer, 2019-08-07:
we are also waiting for a release of both `realsense_camera` and `realsense2_camera`
what's the status here?

airballking, 2019-10-01:
I'd like to +1 the request for a melodic release of the ```realsense2_camera```. Thanks a lot for your support!

KrystianMarek, 2020-01-01:
Hi,

I'm trying to run realsense-ros on a Jetson Nano without much luck, please help.

```
 * /rosdistro: melodic
 * /rosversion: 1.14.3
```
```
$ inxi -F
System:    Host: jetson Kernel: 4.9.140-tegra aarch64 bits: 32 Console: tty 0 Distro: Ubuntu 18.04.3 LTS
Machine:   No /sys/class/dmi; using dmidecode: root required for dmidecode
CPU:       Quad core ARMv8 rev 1 (v8l) (-MCP-)  (ARM)
           clock speeds: max: 1479 MHz 1: 1132 MHz 2: 1132 MHz 3: 1132 MHz 4: 1132 MHz
Graphics:  Card: Failed to Detect Video Card!
           Display Server: X.org 1.19.6 driver: nvidia tty size: 207x61 Advanced Data: N/A out of X
Audio:     Card-1 tegra-hda driver: tegra-hda Sound: ALSA v: k4.9.140-tegra
           Card-2 tegra-snd-t210ref-mobile-rt565x driver: tegra-snd-t210r
Network:   Card-1: Intel Wireless 8265 / 8275 driver: iwlwifi
           IF: wlan0 state: down mac: 98:af:65:2d:9b:9b
           Card-2: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller driver: r8168
           IF: eth0 state: up speed: 1000 Mbps duplex: full mac: 00:04:4b:e6:8e:22
Drives:    HDD Total Size: NA (-)
           ID-1: /dev/mmcblk0 model: N/A size: 63.9GB
Partition: ID-1: / size: 59G used: 16G (28%) fs: ext4 dev: /dev/mmcblk0p1
           ID-2: swap-1 size: 0.52GB used: 0.00GB (0%) fs: swap dev: /dev/zram0
           ID-3: swap-2 size: 0.52GB used: 0.00GB (0%) fs: swap dev: /dev/zram1
           ID-4: swap-3 size: 0.52GB used: 0.00GB (0%) fs: swap dev: /dev/zram2
           ID-5: swap-4 size: 0.52GB used: 0.00GB (0%) fs: swap dev: /dev/zram3
Sensors:   System Temperatures: cpu: 38.0C mobo: N/A
           Fan Speeds (in rpm): cpu: N/A
Info:      Processes: 312 Uptime: 2:36 Memory: 1625.3/3964.3MB Init: systemd runlevel: 5
           Client: Shell (bash) inxi: 2.3.56
```
I was able to install https://github.com/IntelRealSense/librealsense 
the realsense-viewer application works.

When I try to run 
`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

I get:
````
$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to /home/mentis/.ros/log/0eb5abc0-2cb2-11ea-b52a-00044be68e22/roslaunch-jetson-9654.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jetson:37869/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
[... 221 lines cut ...]
[ INFO] [1577895408.807293884]: insert Infrared to Stereo Module
[ INFO] [1577895408.807343884]: insert Infrared to Stereo Module
[ INFO] [1577895408.807395916]: insert Gyro to Motion Module
[ INFO] [1577895408.807444563]: insert Accel to Motion Module
[ INFO] [1577895409.166522686]: SELECTED BASE:Depth, 0
 01/01 17:16:49,228 WARNING [547289534848] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ INFO] [1577895409.305912851]: RealSense Node Is Up!
 01/01 17:16:49,337 WARNING [547289534848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 17:16:49,398 WARNING [547289534848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 17:16:49,520 WARNING [547289534848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 17:16:49,581 WARNING [547289534848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 17:16:49,805 WARNING [547289534848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ERROR] [1577895410.222891005]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
[ERROR] [1577895410.749189374]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
[ERROR] [1577895431.213645061]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
^C 01/01 17:17:14,549 WARNING [547180495232] (sensor.cpp:328) Frame received with streaming inactive,Color0, Arrived,0.000000 1577895434549.132568
terminate called without an active exception
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

berktepebag, 2020-01-06:
@KrystianMarek have you checked this JetsonHacks video? [https://www.youtube.com/watch?v=lL3zxwN5Lnw](url) I don't have the camera so I did not have a chance to check if it is working but if Jim made a video about it, probably it's working.

KrystianMarek, 2020-01-06:
@berktepebag, Yes, I saw the video. It is about installing the realsense camera on Jetson Nano. As I wrote above. I already installed the camera on my Jetson Nano and realsense-viewer works. The problem is with realsense-ros. The camera does not work under ROS.

berktepebag, 2020-01-07:
@KrystianMarek sorry my bad, I was focused on Jetson part of the question. Totally missed the ROS part.

drfg0119, 2020-01-07:
Is there any application on TX2 ? ...I'm really tired about surveying the version matching problem. I'm Melodic/Ubuntu 18.04 with Jetpack 4.2

RealSenseCustomerSupport, 2020-01-31:
----------------------------------------------

Hi @doisyg,

The PR has been submitted recently to address this matter:
https://github.com/ros/rosdistro/pull/23493
Please stay tuned for further updates regarding official binary packages availability.

Thank you!

christian-rauch, 2020-03-05:
The package has been released. This can be closed @doisyg @RealSenseCustomerSupport .

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

@christian-rauch,

Thank you for your feedback!

closed 2020-03-05 by RealSenseCustomerSupport