realsenseai/realsense-ros #387: Installation of ROS wrapper on Raspbian OS

ansuman87, 2018-05-29:
Hi,

If I want to install the ROS wrapper on Raspbian, instead of Ubuntu, how should I proceed? I am intending to use an Intel Realsense D415 camera on an RPi 3 board.
Thank you.

/ansuman

GuidoBartoli, 2018-06-06:
Last week I tried to do that on a RPi3B+ board hoping that the camera could work also on USB 2.0, I followed the standard Linux procedure on Raspbian Stretch, but Realsense Drivers cannot compiled on the ARM architecture.

I hope a workaround can be found, I would be very interested in it...

ansuman87, 2018-06-12:
Did you try the instructions mentioned in this link:
https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

Right now I am following these instructions to install the SDK on Rpi3 with Ubuntu mate on it.

GuidoBartoli, 2018-06-12:
Nice suggestion, I tried this morning to follow exactly those steps (on Rpi3B not plus), everything went fine until compilation step explained [here](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md) where the process gets forever stuck here at `context.cpp`:
```
[ 14%] Building CXX object CMakeFiles/realsense2.dir/src/environment.cpp.o
[ 15%] Building CXX object CMakeFiles/realsense2.dir/src/device_hub.cpp.o
[ 15%] Building CXX object CMakeFiles/realsense2.dir/src/pipeline.cpp.o
[ 16%] Building CXX object CMakeFiles/realsense2.dir/src/archive.cpp.o
[ 17%] Building CXX object CMakeFiles/realsense2.dir/src/context.cpp.o
```
The system seems not stuck, because I can press `CTRL+C`, stop it and resume it, but it cannot go beyond that file.

Anyone else is having a similar issue?

ansuman87, 2018-06-14:
It worked for me on Rpi B+. I followed the instructions here : https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md.
But while building from source, I used the CMake flag "-DFORCE_LIBUVC=true". Follow this link: https://github.com/IntelRealSense/librealsense/issues/1132.
It worked for me. It got stuck at 14% the first time. The second time it seemed to get stuck at 18%, but after a while it continued and finished. The first time, I had a browser running in the background. It might be RAM usage probably, that it got stuck. I don't know for sure. However, it is successfully installed now.
Thanks for your input anyways. :)

ansuman87, 2018-06-15:
I just wanted to clarify that I am using Rpi3 B+.

GuidoBartoli, 2018-06-27:
Ok, I’ll try that, thanks!

closed 2019-03-27 by RealSenseCustomerSupport