realsenseai/realsense-ros #388: Build issue on ROS buildfarm

tfoote, 2018-05-30:
The librealsense if failing to install on the ROS buildfarm when trying to build and installing librealsense.

The error can be seen here: http://build.ros.org/view/Ibin_arm_uThf/job/Ibin_arm_uThf__realsense_camera__ubuntu_trusty_armhf__binary/56/consoleFull
```
02:04:22 Setting up ros-indigo-class-loader (0.3.9-0trusty-20180318-130129-0800) ...
02:04:22 Setting up ros-indigo-librealsense (1.12.1-1trusty-20170329-171541-0700) ...
02:04:24 libkmod: ERROR ../libkmod/libkmod.c:556 kmod_search_moddep: could not open moddep file '/lib/modules/4.15.0-20-generic/modules.dep.bin'
02:04:24 modinfo: ERROR: Module alias uvcvideo not found.
02:04:24 dpkg: error processing package ros-indigo-librealsense (--configure):
02:04:24  subprocess installed post-installation script returned error exit status 1
02:04:24 Setting up ros-indigo-roslint (0.10.0-0trusty-20160323-112711-0700) ...
```

We have recently upgraded the build farm to use a newer kernel (4.15) due to spectre/meltdown workarounds: https://github.com/ros-infrastructure/ros_buildfarm/issues/535 which looks to be a compatibility issue with librealsense.

Clearly librealsense is trying to install a kernel module. Is this something that can scale across versions or have conditional logic to handle a different kernel version. Also since this isn't going to be running on the buildfarm machines, it should be able to build against the older Trusty kernel and be installed there, but the build runs using the newer kernel.

k-okada, 2018-06-05:
@tfoote I ran into same problem, and I thought this is because of missing `udevadm` program, and add them to package.xml (https://github.com/IntelRealSense/librealsense/pull/1829).

But if I compare success  build (http://build.ros.org/view/Ksrc_uX/job/Kbin_uX64__realsense_camera__ubuntu_xenial_amd64__binary/71/consoleFull) and failed build (http://build.ros.org/view/Ksrc_uX/job/Kbin_uX64__realsense_camera__ubuntu_xenial_amd64__binary/92/consoleFull)
Formar one outputs 
```
08:21:15 Setting up ros-kinetic-librealsense (1.12.1-0xenial-20180222-192514-0800) ...
08:21:15 INFO: Docker environment not supported for udevadm; skipping.
08:21:15 INFO: Docker environment not supported for DKMS; skipping.
08:21:15 Setting up ros-kinetic-roslint (0.11.0-0xenial-20180222-193217-0800) .
```
(c.f. https://github.com/intel-ros/librealsense-release/blob/debian/kinetic/xenial/librealsense/debian/postinst#L34-L41 )

This is very strange because, i remember the recent upgrade of the buildfirm changed to use docker...

tfoote, 2018-06-05:
Hmm that's oddly different. Maybe the newer kernel doesn't have the same proc layout and the docker detection is failing.

edowson, 2018-08-06:
I managed to get librealsense to install inside a docker container.

The solution required
- that the kernel versions in the host and container were matched,
- install librealsense uvcvideo kernel module on the host (see `setup.sh`),
- elevate the privileges of the container (see `run.sh`).

Run this `setup.sh` script first on the host:
```
#!/bin/sh

# setup keys for intel's repository server
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
sudo bash -c 'echo "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo $(lsb_release -cs) main" > /etc/apt/sources.list.d/realsense-public.list'
sudo apt-get update

# install linux-kernel-headers on host
sudo apt-get install linux-headers-$(uname -r)

# install librealsense2
sudo apt-get install librealsense2
sudo apt-get install librealsense2-dkms
sudo apt-get install librealsense2-utils
sudo apt-get install realsense-uvcvideo

# add current user to video group
sudo usermod -a -G video $USER
getent group video
```

Prepare a `build.sh` script:
```
#!/bin/sh
docker build \
  --file Dockerfile \
  --build-arg repository=ros \
  --build-arg tag=kinetic-desktop-full-xenial \
  -t ros:kinetic-turtlebot-gazebo7-xenial .
```

Use the following `Dockerfile`. I based this off an NVIDIA OpenGL dockerfile which installed ros-kinetic-core and ros-kinetic-desktop-full, so you'll have to remove the --runtime=nvidia option if you're basing it off the osrf/ros images or create your own ros/opengl image based off NVIDIA's dockerfile.

The trick was ensuring that the host and container kernel versions matched, so that when the container modprobes the uvcvideo module, it can load it when running a container with elevated privileges and appropriate volume mappings to /dev and /lib/modules on the host.
```
# ros:kinetic-turtlebot-gazebo7-xenial

# docker build -t ros:kinetic-turtlebot-gazebo7-xenial .
# xhost +local:root
# docker run --runtime=nvidia -ti --rm -e DISPLAY -e QT_GRAPHICSSYSTEM=native -e QT_X11_NO_MITSHM=1 -v /dev:/dev -v /lib/modules:/lib/modules -v /tmp/.X11-unix:/tmp/.X11-unix --privileged --cap-add=ALL ros:kinetic-turtlebot-gazebo7-xenial
# xhost -local:root

ARG repository
ARG tag
FROM ${repository}:${tag}
LABEL maintainer "Elvis Dowson <elvis.dowson@gmail.com>"
LABEL description "Kinetic ROS Turtlebot with Gazebo 7 and NVIDIA OpenGL - Ubuntu-16.04."

# setup environment
ENV container docker
#ARG DEBIAN_FRONTEND=noninteractive
#ARG TERM=linux

# setup sources.list
RUN echo "deb-src http://us.archive.ubuntu.com/ubuntu/ $(lsb_release -cs) main restricted \n\
deb-src http://us.archive.ubuntu.com/ubuntu/ $(lsb_release -cs)-updates main restricted \n\
deb-src http://us.archive.ubuntu.com/ubuntu/ $(lsb_release -cs)-backports main restricted universe multiverse \n\
deb-src http://security.ubuntu.com/ubuntu $(lsb_release -cs)-security main restricted" \
    > /etc/apt/sources.list.d/official-source-repositories.list

# install kernel image and kernel header packages to match that of the host
RUN apt-get update \
    && apt-get install --no-install-recommends -q -y \
    apt-utils \
    pkg-config \
    && apt-get install --no-install-recommends -q -y \
    linux-image-$(uname -r) \
    linux-headers-$(uname -r) \
    && apt-get autoremove -y \
    && rm -rf /var/lib/apt/lists/*

# setup keys for intel's repository server
RUN apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
RUN echo "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo $(lsb_release -cs) main" \
    > /etc/apt/sources.list.d/realsense-public.list

# install intel realsense libraries
RUN apt-get update \
    && apt-get install --no-install-recommends -q -y \
    librealsense2 \
    librealsense2-dkms \
    librealsense2-utils \
    realsense-uvcvideo \
    && apt-get autoremove -y \
    && rm -rf /var/lib/apt/lists/*

# # install ros device driver packages
# RUN apt-get update \
#     && DEBIAN_FRONTEND=noninteractive apt-get install -q -y \
#     ros-$ROS_DISTRO-cv-bridge \
#     ros-$ROS_DISTRO-librealsense \
#     ros-$ROS_DISTRO-realsense-camera \
#     && apt-get autoremove -y \
#     && rm -rf /var/lib/apt/lists/*

# # install ros turtlebot packages
# RUN apt-get update \
#     && TERM=xterm apt-get install --no-install-recommends -q -y \
#     ros-$ROS_DISTRO-turtlebot-gazebo \
#     && apt-get autoremove -y \
#     && rm -rf /var/lib/apt/lists/*

ENTRYPOINT ["/ros_entrypoint.sh"]
CMD ["bash"]
#CMD ["roslaunch turtlebot_gazebo turtlebot_world.launch"]
```

Use the following `run.sh` script. Remove the nvidia runtime if you're not using it:
```
#!/bin/sh
xhost +local:root
docker run -it \
  --runtime=nvidia \
  -e DISPLAY \
  -e QT_GRAPHICSSYSTEM=native \
  -e QT_X11_NO_MITSHM=1 \
  -v /dev:/dev \
  -v /lib/modules:/lib/modules \
  -v /tmp/.X11-unix:/tmp/.X11-unix \
  --privileged \
  --cap-add=ALL \
  --rm \
  ros:kinetic-turtlebot-gazebo7-xenial
xhost -local:root
```

This should get you as far as being able to install librealsense2 cleanly on the docker image and performing a modprobe for uvcvideo. I don't have actual hardware so haven't been able to confirm it works, but the container should have full access to the host container's usb ports with this setup.

SteveMacenski, 2018-08-06:
If you're looking for a nice work around (like I was) for installing all this stuff on systems and not going through the pain of librealsense installation, I have a snap package for it set up with docs in the repo below.

https://github.com/SteveMacenski/ros-realsense-d400-snap-pkg

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Close this issue for lack of activity. Please feel free to reopen if the issue is still relevant.

tfoote, 2019-03-20:
This is still an issue that people are having to work around.

doronhi, 2019-03-21:
> This is still an issue that people are having to work around.

It is a librealsense installation issue. not realsense2_camera. I suggest you reopen it in librealsense repository.

closed 2019-03-21 by doronhi