realsenseai/realsense-ros #389: can I use D435 with ROS?

andreymathcanary, 2018-05-31:
It seems that it is not supported by ROS. Am I right?

Tuebel, 2018-06-12:
Yes , you can use it, actually I am using it right at this moment.

As the description of the product page states:

> The Intel® RealSense™ Depth Camera D400-Series uses stereo vision to calculate depth. The D435 is a USB-powered depth camera and consists of a pair of depth sensors, RGB sensor, and infrared projector. It is ideal for makers and developers to add depth perception capability to their prototype development.

So it is clearly a D400-Series product and thereby supported.

andreymathcanary, 2018-06-12:
Thank you!

I have found the installation..

Andrey Melnikov, PhD
Philadelphia, PA, USA
cell:   +1-215-485-8292
https://sites.google.com/site/andreymelnikovhomepage/home
<http://www.math.drexel.edu/~andreymath>

SteveMacenski, 2018-08-12:
If this issue is done, can it be closed?

RatheeshRK, 2018-09-16:
> Thank you! I have found the installation.. Andrey Melnikov, PhD Philadelphia, PA, USA cell: +1-215-485-8292 https://sites.google.com/site/andreymelnikovhomepage/home <http://www.math.drexel.edu/~andreymath>
> […](#)

I am using ros kinetic, and trying to use D435 Realsense, but not able to launch the file for D435, It would be helpful, if you could share the steps used. The realsense-viewer is working fine..in Ubuntu 16.04 LTS, (With out ROS).

closed 2018-09-17 by andreymathcanary