realsenseai/realsense-ros #391: Error when subscribing to compressedDepth topic

RajathSwaroop, 2018-06-06:
I use realsense D415 with realsense ros package on ros kinetic and when I subscribe to the topic /camera/color/image_raw/compressedDepth I get the following error:

Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).

What does that mean?? the compression on realsense package is not n correct format?

fiorano10, 2018-08-28:
@RajathSwaroop Do you get the same error after updating the `librealsense2` SDK and the camera FW?

RajathSwaroop, 2018-08-28:
Hi that error was due to the compressedDepth msg type not being built. I do not get the error anymore. Also I should have subscribed to /camera/depth/image_raw_rect/compressedDepth

eirikaso, 2019-03-07:
Hi @RajathSwaroop 
I have the same problem. I get this error when trying to display the compressed depth image with rqt_image_view.

> ImageView.callback_image() while trying to convert image from '16UC1' to 'rgb8' an exception was thrown (Image is wrongly formed: step < width * byte_depth * num_channels  or  640 != 640 * 2 * 1)

What do you mean by the "compressedDepth msg type not being built". When I check the topic the type is "sensor_msgs/CompressedImage", so looks to be OK there

I would really appreciate your help!

closed 2019-03-21 by RealSenseCustomerSupport