realsenseai/realsense-ros #395: catkin_make realsense ros wrapper error

wennycooper, 2018-06-13:
Hi,

I tried to use catkin_make to build the realsense ros wrapper but failed.
It said "Intel RealSense SDK 2.0 is missing", but I did installed it.
Here is what I did:

1. Follow the instruction to install realsense SDK 2.0 Debian package
https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md
(and I confirmed the realsense-viewer run ok)

2. Download the ros wrapper and extract to ~/catkin_ws/src/realsense-2.0.3
3. make
```
$ cd ~/catkin_ws
$ catkin_make
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-2.0.3/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at realsense-2.0.3/realsense2_camera/CMakeLists.txt:29 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

CMake Error at realsense-2.0.3/realsense2_camera/CMakeLists.txt:31 (message):
  

  

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases

  

-- Configuring incomplete, errors occurred!
See also "/home/kkuei/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/kkuei/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

$ rosversion -d
kinetic
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.4 LTS"

wennycooper, 2018-06-19:
Hi mustafaxfe,

I followed the tutorial http://wiki.ros.org/realsense_camera/Tutorials/Building_librealsense_from_Sources and checkout the master branch.

Then I got the package name not found...
```
~/catkin_ws/src$ git clone https://github.com/intel-ros/realsense.git
~/catkin_ws/src$ git clone https://github.com/IntelRealSense/librealsense.git
~/catkin_ws/src$ rosdep update
~/catkin_ws/src$ cd ~/catkin_ws
~/catkin_ws$ rosdep -y install --from-paths src --ignore-src
~/catkin_ws$ cd ~/catkin_ws
~/catkin_ws$ catkin_make --pkg librealsense
Base path: /home/kkuei/catkin_ws
Source space: /home/kkuei/catkin_ws/src
Build space: /home/kkuei/catkin_ws/build
Devel space: /home/kkuei/catkin_ws/devel
Install space: /home/kkuei/catkin_ws/install
Packages "librealsense" not found in the workspace
```

Then I checked the package name in ~/catkin_ws/src/librealsense/package.xml and I found it should be "librealsense2". So I issued the command to build it:

```
~/catkin_ws$ catkin_make --pkg librealsense2
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/kkuei/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/kkuei/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:2462: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```

It looks like it's not a catkin package.... any idea??

wennycooper, 2018-06-20:
Hi, 
According to this page  https://github.com/intel-ros/realsense/releases
The latest supported realsense SDK version is "LibRealSense v2.10.2"

So I did:
1. uninstalled my librealsense 2.12.0.
2. download 2.10.2 from https://github.com/IntelRealSense/librealsense/releases/tag/v2.10.2
3. installed librealsense 2.10.2 based on https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
4. installed realsense ros wrapper  based on https://github.com/intel-ros/realsense

Then it looks ok now.

closed 2018-06-20 by wennycooper

XYudong, 2019-09-23:
> Hi mustafaxfe,
>
> I followed the tutorial http://wiki.ros.org/realsense_camera/Tutorials/Building_librealsense_from_Sources and checkout the master branch.
>
> Then I got the package name not found... [...]

I got the same error