realsenseai/realsense-ros #397: D435 change output resolution with rs_rgb.launch

GITSHOHOKU, 2018-06-18:
Hello , I want to use D435 with resolution 1280x720  using rs_rgb.launch .So I change the default resolution from 640x480 to 1280x720 like  
  ```
<arg name="depth_width"         default="1280"/>
  <arg name="depth_height"        default="720"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra1_width"        default="1280"/>
  <arg name="infra1_height"       default="720"/>
  <arg name="enable_infra1"       default="true"/>
```
I  got warning with `[ WARN] [1529306609.361346371]: Reconfigure callback failed with exception get_xu(id=11) failed! Last Error: Input/output error:`
 
I also try to do this in terminal like 
`roslaunch realsense2_camera rs_rgbd.launch depth_width:=1280 depth_height:=720 infra1_width:=1280 infra1_depth:=720 infra2_width:=1280 infra2_depth:=720 color_width:=1280 color_height:=720`
And also I got warning just like above and addition 
```
[ WARN] [1529306604.247768953]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 1, Format: Y8, Width: 1280, Height: 480, FPS: 30
[ WARN] [1529306604.247849834]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Format: Y8, Width: 1280, Height: 480, FPS: 30
```
The first way to change in launch file can make device to generate 1280x720 images but has warning,I want know is there a easy way to change launch resolution without warning?

Thanks

lz89, 2018-06-19:
The optimal resolution for D435 seems to be 848x480. And make sure your device is recognised as a USB3 device by using the sample program realsense-viewer.

GITSHOHOKU, 2018-10-23:
@lz89 
Thank you , I have changed it to the suggest resolution and it worked

closed 2018-10-23 by GITSHOHOKU