realsenseai/realsense-ros #399: Why is it that the depth_image_proc point_cloud_xyz depth image converter is so much more efficient than built in realsense methods?

SteveMacenski, 2018-06-22:
When I run the nodelet chain to get out a point cloud, it takes about 2% more CPU on the process. When I do so through the realsense ros camera, it adds more than 100% load.

fiorano10, 2018-08-28:
@SteveMacenski depth_image_proc might be downsampling the images for generating the point_cloud.

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Close this issue for lack of activity. Please feel free to reopen if the issue is still relevant.

SteveMacenski, 2019-03-20:
Still open needing a response....

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

Hi @SteveMacenski,

just wondering, does it make any difference if built with OPENMP=false?
`cmake ../  -DBUILD_WITH_OPENMP=false -DCMAKE_BUILD_TYPE=release`

    nuc:~/catkin_ws$ roslaunch realsense2_camera rs_rgbd.launch
    skip---

    SUMMARY
    ========
    PARAMETERS
    skip---
     * /camera/realsense2_camera/base_frame_id: camera_link
     * /camera/realsense2_camera/clip_distance: -1.0
     * /camera/realsense2_camera/color_fps: 30
     * /camera/realsense2_camera/color_frame_id: camera_color_frame
     * /camera/realsense2_camera/color_height: 480
     * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
     * /camera/realsense2_camera/color_width: 640
     * /camera/realsense2_camera/depth_fps: 30
     * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
     * /camera/realsense2_camera/depth_height: 480
     * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
     * /camera/realsense2_camera/depth_width: 640
    skip--- * /rosdistro: kinetic
    NODES
      /camera/
        color_rectify_color (nodelet/nodelet)
        points_xyzrgb_hw_registered (nodelet/nodelet)
        realsense2_camera (nodelet/nodelet)
        realsense2_camera_manager (nodelet/nodelet)

    auto-starting new master
    process[master]: started with pid [10484]
    ROS_MASTER_URI=http://localhost:11311

    setting /run_id to 7744057c-4f5b-11e9-ada9-f44d30674271
    process[rosout-1]: started with pid [10497]
    started core service [/rosout]
    process[camera/realsense2_camera_manager-2]: started with pid [10514]
    process[camera/realsense2_camera-3]: started with pid [10515]
    process[camera/color_rectify_color-4]: started with pid [10516]
    process[camera/points_xyzrgb_hw_registered-5]: started with pid [10523]
    [INFO] [1553558987.819203063]: Initializing nodelet with 4 worker threads.
    [INFO] [1553558987.915983968]: RealSense ROS v2.2.1
    [INFO] [1553558987.916052966]: Running with LibRealSense v2.19.1
    [INFO] [1553558988.072408994]: getParameters...
    [INFO] [1553558988.147450091]: setupDevice...
    [INFO] [1553558988.147480889]: JSON file is not provided
    [INFO] [1553558988.147502754]: ROS Node Namespace: camera
    [INFO] [1553558988.147525347]: Device Name: Intel RealSense D415
    skip---
    [INFO] [1553558988.147708502]: Align Depth: On
    [INFO] [1553558988.147744987]: Sync Mode: On
    [INFO] [1553558988.147821089]: Device Sensors:
    [INFO] [1553558988.147873429]: Stereo Module was found.
    [INFO] [1553558988.147894385]: RGB Camera was found.
    skip---

    nuc:~$ top -b -n 1top - 17:15:15 up  1:00,  1 user,  load average: 1.75, 1.31, 1.32
    Tasks: 214 total,  1 running, 212 sleeping,  0 stopped,  1 zombie
    %Cpu(s): 23.9 us,  3.0 sy,  0.1 ni, 72.5 id,  0.1 wa,  0.0 hi,  0.3 si,  0.0 st
    KiB Mem :  3927392 total,  1802072 free,  1159700 used,  965620 buff/cache
    KiB Swap:  4080636 total,  3493372 free,  587264 used.  2205120 avail Mem

      PID       PR  NI    VIRT    RES    SHR S  %CPU %MEM    TIME+ COMMAND
    10514       20  0 1201812  53428  28524 S  18.8  1.4  0:54.38 nodelet
     1079       20  0  437708  55668  46136 S  12.5  1.4  8:57.99 Xorg
     5355       20  0  681608  19616  13560 S  6.2  0.5  2:42.17 gnome-system-mo
    11179       20  0  48872  3640  3052 R  6.2  0.1  0:00.02 top
        1       20  0  185456  3372  2188 S  0.0  0.1  0:01.78 systemd

closed 2019-04-25 by RealSenseCustomerSupport