realsenseai/realsense-ros #401: Nodelet Bond Broken

caiofis, 2018-06-27:
When I try to use the D435 on a system with two computers connected in a network my camera stop publishing after some seconds and give me a "Bond broken" error. When I run it using the local machine as ros master it works all right. Did anyone faced this problem?

closed 2018-07-10 by caiofis

JRauer, 2018-09-18:
I've got exactly the same problem - also with other realsense cameras (such as sr300). @caiofis how did you sort this out?

yangautumn, 2018-12-12:
me too. I am using D435. Wondering the solution.

yangautumn, 2018-12-14:
in the master machine, I added one line **10.0.0.2  tegra-ubuntu** to the **/etc/hosts** file, where **10.0.0.2** is the IP address of the second machine and **tegra-ubuntu** is the hostname or domain of the second machine. By doing that, I have the issue solved.

SCH923, 2019-11-27:
works great thx