realsenseai/realsense-ros #408: Red image

sanjifr3, 2018-07-11:
When using the launch script, and rqt_image_view to view the topic "/camera/color/image_raw", the red in the image is pronounced. Is there a way to fix this?

[image]

dorodnic, 2018-10-15:
Colour camera settings are persistent and stored in camera flash. You can use `realsense-viewer` tool to adjust various colour sensor options.

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Close this issue for lack of activity. Please feel free to reopen if the issue is still relevant.

closed 2019-03-20 by sanjifr3

SteveMacenski, 2019-03-20:
I'd recommend updating the defaults in ROS drivers to not have this red feature. I've seen this across nearly all (several hundred) units I've touched.

doronhi, 2019-03-21:
> I'd recommend updating the defaults in ROS drivers to not have this red feature. I've seen this across nearly all (several hundred) units I've touched.

I think it would be more beneficial to suggest it in the librealsense github.

SteveMacenski, 2019-03-21:
Its not a librealsense issue, the defaults in librealsense are good. Its the defaults in this repo that are not

doronhi, 2019-03-21:
> Its not a librealsense issue, the defaults in librealsense are good. Its the defaults in this repo that are not

I must have misunderstood you. What defaults are you referring to?

SteveMacenski, 2019-03-21:
The white balance. In RS view all is good, its only the ROS realsense driver that produces the red images. I found `rs435_color_white_balance 360` to work pretty well vs the 460 that's default

doronhi, 2019-03-21:
I don't see it, at least not in the latest realsense2_camera versions. 
I agree that 3600 is better than 4600 but that is the device's default. Also, it is, by default, set to automatic white balance so it doesn't matter. 
It's the same with realsense-viewer.  Neither of them change the device's values by default.