realsenseai/realsense-ros #409: DS5 group_devices is empty, rame metadata isn't available! Errors

fiorano10, 2018-07-11:
After updating the SDK, librealsense and the realsense. While launching rs_camera.launch I keep getting these warnings:

[ INFO] [1531352172.037410555]: Initializing nodelet with 8 worker threads.
[ INFO] [1531352172.068490136]: RealSense ROS v2.0.3
[ INFO] [1531352172.068517215]: Running with LibRealSense v2.12.0
 11/07 19:36:12,075 WARNING [140280209618816] (ds5-factory.cpp:289) DS5 group_devices is empty.
[ INFO] [1531352172.326550446]: getParameters...
[ INFO] [1531352172.373278954]: setupDevice...
[ INFO] [1531352172.373306672]: JSON file is not provided
[ INFO] [1531352172.373325950]: ROS Node Namespace: camera
[ INFO] [1531352172.373346899]: Device Name: Intel RealSense D435
[ INFO] [1531352172.373364034]: Device Serial No: 746512070198
[ INFO] [1531352172.373382469]: Device FW version: 05.09.14.00
[ INFO] [1531352172.373402627]: Device Product ID: 0x0B07
[ INFO] [1531352172.373421268]: Enable PointCloud: Off
[ INFO] [1531352172.373438880]: Align Depth: On
[ INFO] [1531352172.373456242]: Sync Mode: On
[ INFO] [1531352172.373632103]: Device Sensors: 
[ INFO] [1531352172.373759879]: Stereo Module was found.
[ INFO] [1531352172.373789097]: RGB Camera was found.
[ INFO] [1531352172.373921944]: setupPublishers...
[ INFO] [1531352172.376573272]: Expected frequency for depth = 15.00000
[ INFO] [1531352172.402793346]: Expected frequency for infra1 = 15.00000
[ INFO] [1531352172.426064998]: Expected frequency for aligned_depth_to_infra1 = 15.00000
[ INFO] [1531352172.449513732]: Expected frequency for infra2 = 15.00000
[ INFO] [1531352172.471717556]: Expected frequency for aligned_depth_to_infra2 = 15.00000
[ INFO] [1531352172.495937390]: setupStreams...
 11/07 19:36:12,496 WARNING [140280209618816] (sensor.cpp:318) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1531352172.498520087]: depth stream is enabled - width: 1280, height: 720, fps: 15
[ INFO] [1531352172.499799330]: infra1 stream is enabled - width: 1280, height: 720, fps: 15
[ INFO] [1531352172.500956731]: infra2 stream is enabled - width: 1280, height: 720, fps: 15
[ INFO] [1531352172.510652725]: publishStaticTransforms...
[ INFO] [1531352172.510921031]: RealSense Node Is Up!
 **11/07 19:36:12,655 WARNING [140279453570816] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1531352172.659197949]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 11/07 19:36:17,642 WARNING [140280209618816] (types.cpp:57) set_xu(id=2) failed! Last Error: Input/output error
[ WARN] [1531352177.643012622]: Reconfigure callback failed with exception set_xu(id=2) failed! Last Error: Input/output error:**

CxyCH, 2018-08-01:
hi: 
   I also got this warning too. Now do you know what's wrong with that and what's the impact to the data with this warning?

fiorano10, 2018-08-01:
@CxyCH, see this [Issue #2043](https://github.com/IntelRealSense/librealsense/issues/2043)

closed 2019-01-15 by fiorano10