realsenseai/realsense-ros #410: [realsense2] release to kinetic/ubuntu 16.04

fmessmer, 2018-07-12:
do you plan to release `realsense2_camera` and `librealsense2` into kinetic/xenial anytime soon?
it'd be highly appreciated....

YuehChuan, 2018-07-14:
can we use D435 camera with ros wrapper? 
Without ros I could run it correctly 
I got such error  when `roslaunch realsense2_camera rs_rgbd.launch`
[image]

Thanks, or I should open new issue.

fiorano10, 2018-08-28:
@fmessmer I've been running `realsense2_camera` and `librealsense2` on Kinetic/Xenial for the past 4 months

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

Close this issue for lack of activity. Please feel free to reopen if the issue is still relevant.

SteveMacenski, 2019-03-21:
Still relevant

closed 2019-03-28 by RealSenseCustomerSupport