realsenseai/realsense-ros #411: Use .json presets files

martinakos, 2018-07-13:
With realsense-viewer I can adjust the D435 parameters and save them as a preset in a .json file, then I can load them using librealsense API. Is there any way of loading these .json preset files in a ROS .launch file? I have already seen [this](http://wiki.ros.org/realsense_camera/Tutorials/change_camera_parameters), but I hope I can use the .json files somehow. 

Or alternatively, is there anyway to make these .json preset files available as options in the Dynamic Reconfigure rs435_depth_visual_preset drop down list?

martinakos, 2018-07-20:
Now I realise that the launch files can contain a tag: 
`<arg name="json_file_path"      default=""/>`
I have set this path to my preset file but it doesn't seem to load it. In fact in the output messages I can see: 
`[ INFO] [1532071455.622040164]: JSON file is not provided`

so it's not reading it. Am I missing something else?

rupak-d, 2018-09-04:
Hi,
Any solution for this?

RealSenseCustomerSupport, 2019-03-28:
----------------------------------------------

Hi martinakos,

You can directly set json file into to a ROS launch file to take effect.

For example, for rs_camera.launch, add absolute path info to a json file to be loaded with here:
https://github.com/intel-ros/realsense/blob/development/realsense2_camera/launch/rs_camera.launch#L3

Thanks!

RealSenseCustomerSupport, 2019-04-09:
----------------------------------------------

Hi martinakos,

Wonder if you have any update? Or if nothing else is needed, this one will be closed soon.

Thanks!

closed 2019-04-19 by RealSenseCustomerSupport