realsenseai/realsense-ros #412: [D435 ]/opt/ros/kinetic/lib/libbondcpp.so: undefined symbol: _ZN3ros11SteadyTimerD1Ev

YuehChuan, 2018-07-14:
camera model D435 
4.13.0-43-generic kernel  ubuntu16.04.3
ROS kinetic 
package:
realsense2_camera

I get apt ros-kinetic-nodelet  ros-kinetic-nodelet-core installed

Without ros I could run it correctly.
I got such error when roslaunch realsense2_camera rs_rgbd.launch
```bash=
[ INFO] [1531594763.160792828]: Initializing nodelet with 8 worker threads.
/opt/ros/kinetic/lib/nodelet/nodelet: symbol lookup error: /opt/ros/kinetic/lib/libbondcpp.so: undefined symbol: _ZN3ros11SteadyTimer4stopEv
[FATAL] [1531594763.227766803]: Failed to load nodelet '/camera/points_xyzrgb_hw_registered` of type `depth_image_proc/point_cloud_xyzrgb` to manager `realsense2_camera_manager'
[FATAL] [1531594763.227823744]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
```

[image]

relate issue:
https://answers.ros.org/question/225986/nodelet-service-failed-on-symbol-lookup/
https://github.com/intel-ros/realsense/issues/343

edwardsteds, 2018-09-15:
Same issue ..

qiupeizhao, 2018-09-21:
Same issue. Someone please answer

zwzzsyxqq, 2018-09-28:
Same issue!

Abhicoder1999, 2018-10-03:
same issue!!

Williamc17, 2019-01-13:
same issue!!!

doronhi, 2019-01-15:
It's not happening for me.
@YuehChuan , what do you mean by "Without ros I could run it correctly."?
Do you manage to reproduce the error with an updated version of librealsense (2.17.1)?
I assume the realsense_2_camera is up to date (version 2.1.3)

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

Hi @YuehChuan,
Does the issue persist with the latest ROS wrapper (2.2.1) and librealsense (2.19.x) combo?

YuehChuan, 2019-04-01:
@RealSenseCustomerSupport  
Thanks for pinging me, I have no hardware for now, so cannot test it.

closed 2019-04-09 by RealSenseCustomerSupport

xtywbpute, 2019-09-25:
> camera model D435
> 4.13.0-43-generic kernel ubuntu16.04.3
> ROS kinetic
> D435 Firmware :05.09.09.02
> librealsense build from source: 2.13.0
> 
> package:
> realsense2_camera
> 
> I get apt ros-kinetic-nodelet ros-kinetic-nodelet-core installed
> 
> Without ros I could run it correctly.
> I got such error when roslaunch realsense2_camera rs_rgbd.launch
> 
> ```
> [ INFO] [1531594763.160792828]: Initializing nodelet with 8 worker threads.
> /opt/ros/kinetic/lib/nodelet/nodelet: symbol lookup error: /opt/ros/kinetic/lib/libbondcpp.so: undefined symbol: _ZN3ros11SteadyTimer4stopEv
> [FATAL] [1531594763.227766803]: Failed to load nodelet '/camera/points_xyzrgb_hw_registered` of type `depth_image_proc/point_cloud_xyzrgb` to manager `realsense2_camera_manager'
> [FATAL] [1531594763.227823744]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
> ```
> 
> [image]
> 
> relate issue:
> https://answers.ros.org/question/225986/nodelet-service-failed-on-symbol-lookup/
> #343
> #349
> #410

Please upgrade the [indigo] of catkin_workspace package [ros_comm]  to [kinetic]

ChaeChae0505, 2022-01-26:
please answer,,,,,,

XiaoJake, 2022-02-11:
same issue!
But, D435 can work fine on my other Xavier, very strange.
I don't know what the problem is when I build the new system environment.