realsenseai/realsense-ros #413: Setting Depth Clamp Min/Max in intel-ros

arrfou90, 2018-07-16:
Hi, setting the Depth Clamp Min/Max for Intel RealSense D400 series cameras is not provided by this ros package. Do you know how I can easily do that through the configuration file.

closed 2018-11-22 by arrfou90

frk1993, 2019-06-03:
Hi,
do you have a solution for this problem ?

best regards

doronhi, 2019-06-04:
There is the flag `clip_distance` as described in the README.md file for clipping points at distance farther then a given value. There is no option for clipping points with shorter distance then a given value .

TiagoTostas, 2020-09-18:
Hi,
About the clipping distance, does the clipping distance affects the resolution of the pointcloud?
Or what is the main advantage compared to the distance threshold filtering?

Thanks

doronhi, 2020-09-20:
The distance threshold filter is not available in the ROS wrapper.
Clipping the distance has no effect on the resolution or anything else. It just remove pixels of range greater than the given value.

TiagoTostas, 2020-09-20:
Thanks
Meanwhile I realised that my question was more about the configurations for min/max range, rather than a problem about the clipping distance