realsenseai/realsense-ros #422: Why do modifier.setPointCloud2Fields and modifier.setPointCloud2FieldsByString at the same time?

testkit, 2018-07-20:
This might not be an issue but a question. 
When I look at src/base_realsense_node.cpp line 1062~1069, I noticed that modifier.setPointCloud2Fields and modifier.setPointCloud2FieldsByString are called in sequence. But seems like point_step, row_step, and data.resize will be recalculated by modifier.setPointCloud2FieldsByString, and I didn't see any impact in rviz if I comment out modifier.setPointCloud2Fields. 
Anyone can help me out there to let me know what's the usage to call modifier.setPointCloud2Fields beforehand, or it's just a piece of redundant code? 
Thanks!

testkit, 2018-08-10:
Anyone has answer?

RealSenseCustomerSupport, 2019-04-22:
----------------------------------------------

Hi testkit,

Sorry for coming to this one late. And thanks for the feedback.
ROS 2.1.3 release has address the issue.

Thanks!

closed 2019-04-29 by RealSenseCustomerSupport