realsenseai/realsense-ros #424: CPU issues rs_camera.launch

Bwaki, 2018-07-23:
Hello everyone,

Before starting to explain my issues i apologies if i forget details or important informations about my setup if you need more informations about my configuration, don't hesitate to ask me.

I work with : 
- Intel Realsense D435 camera (USB 3.0)
    - Firmware version : 05.09.14.00 (last version available normally)
- Kernel version : 4.13.0
- Ubuntu 16.04
- ROS Kinetic 
- realsense package version : 1.1.2.realsense-1.3.0 (from this command : modinfo uvcvideo | grep "version:")
- librealsense (last version available on github)
- Intel® Core™ i7-6700HQ CPU 
- 64 bits

OK now my issues is that my CPU go to 100% just by launching : roslaunch realsense2_camera rs_camera.launch 

here is the output of this command : 

```
... logging to /home/guerin/.ros/log/033ecc6a-8e69-11e8-adb4-001696e92150/roslaunch-guerin-Skylake-15881.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.8:33095/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.14.2

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [16504]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 033ecc6a-8e69-11e8-adb4-001696e92150
process[rosout-1]: started with pid [16959]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [16962]
process[camera/realsense2_camera-3]: started with pid [16967]
[ INFO] [1532344232.420600867]: Initializing nodelet with 8 worker threads.
[ INFO] [1532344232.505491000]: RealSense ROS v2.0.3
[ INFO] [1532344232.505619349]: Running with LibRealSense v2.14.0
[ INFO] [1532344232.699186610]: getParameters...
[ INFO] [1532344232.725397718]: setupDevice...
[ INFO] [1532344232.725462270]: JSON file is not provided
[ INFO] [1532344232.725496251]: ROS Node Namespace: camera
[ INFO] [1532344232.725544079]: Device Name: Intel RealSense D435
[ INFO] [1532344232.725574133]: Device Serial No: 819112071970
[ INFO] [1532344232.725620671]: Device FW version: 05.09.14.00
[ INFO] [1532344232.725650983]: Device Product ID: 0x0B07
[ INFO] [1532344232.725676883]: Enable PointCloud: Off
[ INFO] [1532344232.725707892]: Align Depth: Off
[ INFO] [1532344232.725779356]: Sync Mode: Off
[ INFO] [1532344232.726118754]: Device Sensors: 
[ INFO] [1532344232.726281193]: Stereo Module was found.
[ INFO] [1532344232.726371202]: RGB Camera was found.
[ INFO] [1532344232.726592950]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1532344232.726652668]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1532344232.726695063]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1532344232.726763987]: setupPublishers...
[ INFO] [1532344232.731662259]: Expected frequency for depth = 30.00000
[ INFO] [1532344232.762985663]: Expected frequency for infra1 = 30.00000
[ INFO] [1532344232.786148562]: Expected frequency for infra2 = 30.00000
[ INFO] [1532344232.810989797]: Expected frequency for color = 30.00000
[ INFO] [1532344232.834866007]: setupStreams...
 23/07 13:10:32,837 WARNING [140152698521472] (sensor.cpp:334) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1532344232.843092883]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1532344232.843504698]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1532344232.843790462]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1532344232.855195104]: color stream is enabled - width: 640, height: 480, fps: 30
 23/07 13:10:32,875 WARNING [140151887193856] (backend-v4l2.cpp:870) Empty frame has arrived.
 23/07 13:10:32,875 WARNING [140151887193856] (backend-v4l2.cpp:870) Empty frame has arrived.
 23/07 13:10:32,875 WARNING [140151887193856] (backend-v4l2.cpp:870) Empty frame has arrived.
 23/07 13:10:32,875 WARNING [140151887193856] (backend-v4l2.cpp:870) Empty frame has arrived.
[ INFO] [1532344232.879527765]: publishStaticTransforms...
[ INFO] [1532344232.880257498]: RealSense Node Is Up!
 23/07 13:10:32,881 WARNING [140151876327168] (backend-v4l2.cpp:870) Empty frame has arrived.
 23/07 13:10:32,881 WARNING [140151876327168] (backend-v4l2.cpp:870) Empty frame has arrived.
 23/07 13:10:32,881 WARNING [140151876327168] (backend-v4l2.cpp:870) Empty frame has arrived.
 23/07 13:10:32,881 WARNING [140151876327168] (backend-v4l2.cpp:870) Empty frame has arrived.

```

closed 2018-07-24 by Bwaki

fiorano10, 2018-08-08:
@Bwaki how did you solve the `Reconfigure callback failed` issue?

Bwaki, 2018-08-09:
Hi @fiorano10 , my problem was that i was running 2 roscore a the same time, the normal one and a modify version but the issues is that the modify version was using all my cpu so i just remove it and all is back to normal.
Yeah it is a dumb error ^^"