realsenseai/realsense-ros #431: Negative frame latency.

pavloblindnology, 2018-08-02:
Using D435 camera (firmware 5.9.14, librealsense 2.14.0) with ROS Kinetic (realsense2_camera 2.0.3)
I often get negative latency with command
`rostopic delay /camera/infra1/image_rect_raw`.

pavloblindnology, 2018-08-02:
It seems to be related to #419.
According to the latter, we get ~0 latency for the 1st frame and, if there is no time drift between camera and host clocks, for the next ones. And this obviously can not be true, because real latency = up to 1/FPS ms + OS pipe latency.
And the solution is the same as in #419 - to have a function to get current camera time. Then we could synchronize camera and host clocks - of course, up to the function latency.

shuntaraw, 2018-09-10:
I think this is related to #461. The reason for this is that on-device clock is inaccurate and almost always gains.

pavloblindnology, 2018-09-11:
@shuntaraw Yes, for sure. And also because current synchronization is done stupidly, giving 0 latency for the 1st frame, that leads to later latencies fluctuating around 0 even if there were no drift.

closed 2019-04-19 by RealSenseCustomerSupport