realsenseai/realsense-ros #432: Can't work with multi D435 camera

CxyCH, 2018-08-02:
hi:
   My system is ubuntu 16.04 and kernel is 4.15, I install the librealsense  Debian Package, I connect two D435 to PC directly,  without using hub.
   When I try to get images from one camera everything works fine;
   When I try to get images from two cameras sometimes works fine but most of the times give the following:

`[ INFO] [1533215772.233823823]: Initializing nodelet with 8 worker threads.
[ INFO] [1533215772.243803431]: Initializing nodelet with 8 worker threads.
[ INFO] [1533215772.254838221]: RealSense ROS v2.0.3
[ INFO] [1533215772.254866547]: Running with LibRealSense v2.14.1
[ INFO] [1533215772.280641966]: RealSense ROS v2.0.3
[ INFO] [1533215772.280672359]: Running with LibRealSense v2.14.1
[FATAL] [1533215772.467921843]: The requested device with serial number serial_no_camera2 is NOT found!
[FATAL] [1533215772.468226067]: The requested device with serial number serial_no_camera1 is NOT found!
[FATAL] [1533215772.660861554]: Failed to load nodelet '/camera2/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[FATAL] [1533215772.662714723]: Failed to load nodelet '/camera1/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera1/realsense2_camera_manager-2] process has died [pid 26948, exit code 1, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/cxy/.ros/log/39e312c2-9656-11e8-a384-a434d994d108/camera1-realsense2_camera_manager-2.log].
log file: /home/cxy/.ros/log/39e312c2-9656-11e8-a384-a434d994d108/camera1-realsense2_camera_manager-2*.log
[camera2/realsense2_camera_manager-4] process has died [pid 26950, exit code 1, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/cxy/.ros/log/39e312c2-9656-11e8-a384-a434d994d108/camera2-realsense2_camera_manager-4.log].
log file: /home/cxy/.ros/log/39e312c2-9656-11e8-a384-a434d994d108/camera2-realsense2_camera_manager-4*.log
[camera1/realsense2_camera-3] process has died [pid 26949, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/cxy/.ros/log/39e312c2-9656-11e8-a384-a434d994d108/camera1-realsense2_camera-3.log].
log file: /home/cxy/.ros/log/39e312c2-9656-11e8-a384-a434d994d108/camera1-realsense2_camera-3*.log
[camera2/realsense2_camera-5] process has died [pid 26951, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/cxy/.ros/log/39e312c2-9656-11e8-a384-a434d994d108/camera2-realsense2_camera-5.log].
log file: /home/cxy/.ros/log/39e312c2-9656-11e8-a384-a434d994d108/camera2-realsense2_camera-5*.log
`
    I try to run this two D435 in windows 10 and it can work  fine, but  I run realsense-viewer in ubuntu 16.04, only one camera can work fine in the same time .

Thanks in advance for helping me

CxyCH, 2018-08-04:
I fix that problem by source install, but when I run the `roslaunch realsense2_camera rs_multiple_devices.launch`, I get some warning as following:
`[140162391607168] (backend-v4l2.cpp:1098) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 04/08 10:51:27,851 WARNING [140162391607168] (sensor.cpp:313) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1533351087.856460797]: color stream is enabled - width: 640, height: 480, fps: 30
 04/08 10:51:27,922 WARNING [140556067915520] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1533351087.924696876]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
[ INFO] [1533351087.939892093]: publishStaticTransforms...
[ INFO] [1533351087.941409190]: RealSense Node Is Up!
 04/08 10:51:28,046 WARNING [140161595864832] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1533351088.048975698]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 04/08 10:51:28,287 WARNING [140556027668224] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
 04/08 10:51:28,331 ERROR [140556834178944] (types.h:178) xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range
[ WARN] [1533351088.331343572]: Reconfigure callback failed with exception xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range: 
 04/08 10:51:28,411 WARNING [140161584998144] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
 04/08 10:51:28,463 ERROR [140162391607168] (types.h:178) xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range
`
So what wrong for me at this time?

freemanlo, 2018-08-17:
Hi @CxyCH,

Please install the DKMS package to patch above metadata warning.
$ sudo apt-get install librealsense2-dkms

Refer to below page:
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

doronhi, 2018-08-22:
Hi @CxyCH,
Also notice that after downloading the latest version in the repository you can run the nodes for each camera separately, in different terminals and under different namespaces using the parameter "camera" like so:

roslaunch realsense2_camera rs_rgbd.launch camera:=**my_cam1**

now the topics have a distinct header:
/**my_cam1**/aligned_depth_to_color/image_raw

closed 2018-09-03 by doronhi

salihmarangoz, 2020-08-28:
Solved this issue with timed_roslaunch. I have tested multiple D435's but starting all at once failed. Instead starting them one by one worked well. Good luck.
http://wiki.ros.org/timed_roslaunch