realsenseai/realsense-ros #433: Error with realsense2_camera rs_rgbd.launch command

michaelchi08, 2018-08-04:
roslaunch realsense2_camera rs_rgbd.launch
... logging to /home/michael/.ros/log/d598f7f0-979f-11e8-92d9-f894c28abe6b/roslaunch-michael-Precision-7520-8146.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://michael-Precision-7520:43413/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/camera_aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /camera/
    color_rectify_color (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [8158]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d598f7f0-979f-11e8-92d9-f894c28abe6b
process[rosout-1]: started with pid [8172]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [8189]
process[camera/realsense2_camera-3]: started with pid [8190]
process[camera/color_rectify_color-4]: started with pid [8191]
process[camera/points_xyzrgb_hw_registered-5]: started with pid [8198]
[ INFO] [1533357337.925366455]: Initializing nodelet with 8 worker threads.
[ INFO] [1533357337.968521708]: RealSense ROS v2.0.3
[ INFO] [1533357337.968546297]: Running with LibRealSense v2.10.2
[ INFO] [1533357338.164969320]: getParameters...
[ INFO] [1533357338.192757347]: setupDevice...
[ INFO] [1533357338.192780289]: JSON file is not provided
[ INFO] [1533357338.192815694]: ROS Node Namespace: camera
[ INFO] [1533357338.192842222]: Device Name: Intel RealSense 435
[ INFO] [1533357338.192857551]: Device Serial No: 752112071050
[ INFO] [1533357338.192872658]: Device FW version: 05.09.14.00
[ INFO] [1533357338.192887737]: Device Product ID: 0x0B07
[ INFO] [1533357338.192900291]: Enable PointCloud: Off
[ INFO] [1533357338.192912933]: Align Depth: On
[ INFO] [1533357338.192924869]: Sync Mode: On
[ INFO] [1533357338.192983828]: Device Sensors: 
[ INFO] [1533357338.193011851]: Stereo Module was found.
[ INFO] [1533357338.193026957]: RGB Camera was found.
[ INFO] [1533357338.193058153]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1533357338.193071760]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1533357338.193084553]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1533357338.193099458]: setupPublishers...
[ INFO] [1533357338.195341825]: Expected frequency for depth = 30.00000
[ INFO] [1533357338.217421732]: Expected frequency for infra1 = 30.00000
[ INFO] [1533357338.237357933]: Expected frequency for aligned_depth_to_infra1 = 30.00000
[ INFO] [1533357338.255460410]: Expected frequency for infra2 = 30.00000
[ INFO] [1533357338.275718896]: Expected frequency for aligned_depth_to_infra2 = 30.00000
[ INFO] [1533357338.296067153]: Expected frequency for color = 30.00000
[ INFO] [1533357338.315663090]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1533357338.338173615]: setupStreams...
 04/08 00:35:38,339 WARNING [140608344901504] (sensor.cpp:313) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1533357338.341838731]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1533357338.342214961]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1533357338.342568617]: infra2 stream is enabled - width: 640, height: 480, fps: 30
 04/08 00:35:38,344 WARNING [140608344901504] (backend-v4l2.cpp:1098) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 04/08 00:35:38,344 WARNING [140608344901504] (sensor.cpp:313) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1533357338.349724377]: color stream is enabled - width: 640, height: 480, fps: 30
 04/08 00:35:38,364 WARNING [140607545923328] (backend-v4l2.cpp:808) Empty frame has arrived.
 04/08 00:35:38,364 WARNING [140607545923328] (backend-v4l2.cpp:808) Empty frame has arrived.
 04/08 00:35:38,364 WARNING [140607545923328] (backend-v4l2.cpp:808) Empty frame has arrived.
 04/08 00:35:38,364 WARNING [140607545923328] (backend-v4l2.cpp:808) Empty frame has arrived.
[ INFO] [1533357338.366918113]: publishStaticTransforms...
[ INFO] [1533357338.367022663]: RealSense Node Is Up!
 04/08 00:35:38,369 WARNING [140607535056640] (backend-v4l2.cpp:808) Empty frame has arrived.
 04/08 00:35:38,369 WARNING [140607535056640] (backend-v4l2.cpp:808) Empty frame has arrived.
 04/08 00:35:38,369 WARNING [140607535056640] (backend-v4l2.cpp:808) Empty frame has arrived.
 **04/08 00:35:38,369 WARNING [140607535056640] (backend-v4l2.cpp:808) Empty frame has arrived.
 04/08 00:35:38,546 WARNING [140607545923328] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
**[ WARN] [1533357338.548865934]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 04/08 00:35:38,560 WARNING [140607535056640] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
 04/08 00:35:43,395 WARNING [140608344901504] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error
[ WARN] [1533357343.395469281]: Reconfigure callback failed with exception get_xu(id=11) failed! Last Error: Input/output error:**** 

Could you please help? Thank you!

doronhi, 2018-08-22:
I am having trouble recreating your problem.
What OS do you use?
Are you able to see video using realsense-viewer?
I see that you use LibRealSense v2.10.2. Can you upgrade?

jmichaux, 2018-09-10:
I'm having the same error after running `roslaunch realsense2_camera rs_camera.launch` :

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/camera_aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://011504P0002.local:11311

process[camera/realsense2_camera_manager-1]: started with pid [5782]
process[camera/realsense2_camera-2]: started with pid [5783]
[ INFO] [1536599553.407643980]: Initializing nodelet with 4 worker threads.
[ INFO] [1536599553.438722127]: RealSense ROS v2.0.3
[ INFO] [1536599553.438753209]: Running with LibRealSense v2.14.0
[ INFO] [1536599553.622145134]: getParameters...
[ INFO] [1536599553.711585128]: setupDevice...
[ INFO] [1536599553.711640581]: JSON file is not provided
[ INFO] [1536599553.711679625]: ROS Node Namespace: camera
[ INFO] [1536599553.711718041]: Device Name: Intel RealSense D435
[ INFO] [1536599553.711746388]: Device Serial No: 817412070743
[ INFO] [1536599553.711776230]: Device FW version: 05.09.14.00
[ INFO] [1536599553.711806232]: Device Product ID: 0x0B07
[ INFO] [1536599553.711832104]: Enable PointCloud: Off
[ INFO] [1536599553.711861256]: Align Depth: Off
[ INFO] [1536599553.711896535]: Sync Mode: Off
[ INFO] [1536599553.712173844]: Device Sensors: 
[ INFO] [1536599553.712373897]: Stereo Module was found.
[ INFO] [1536599553.712428746]: RGB Camera was found.
[ INFO] [1536599553.712660447]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1536599553.712713122]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1536599553.712745610]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1536599553.712884491]: setupPublishers...
[ INFO] [1536599553.718852401]: Expected frequency for depth = 30.00000
[ INFO] [1536599553.776155918]: Expected frequency for infra1 = 30.00000
[ INFO] [1536599553.832234595]: Expected frequency for infra2 = 30.00000
[ INFO] [1536599553.890012193]: Expected frequency for color = 30.00000
[ INFO] [1536599553.952366849]: setupStreams...
 10/09 12:12:33,953 WARNING [140636692191104] (sensor.cpp:334) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1536599553.954823980]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1536599553.955227545]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1536599553.955559626]: infra2 stream is enabled - width: 640, height: 480, fps: 30
 10/09 12:12:33,956 WARNING [140636692191104] (backend-v4l2.cpp:1160) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 10/09 12:12:33,956 WARNING [140636692191104] (sensor.cpp:334) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1536599553.959130007]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1536599553.970511344]: publishStaticTransforms...
[ INFO] [1536599553.970961620]: RealSense Node Is Up!
 10/09 12:12:34,162 WARNING [140636036695808] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1536599554.163508347]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
[ WARN] [1536599554.163625744]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 10/09 12:12:34,168 WARNING [140635928364800] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
 10/09 12:12:39,040 WARNING [140636692191104] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error
[ WARN] [1536599559.040692507]: Reconfigure callback failed with exception get_xu(id=11) failed! Last Error: Input/output error:

dan9thsense, 2018-10-14:
Exact same problem here.  Any suggestions?

jmichaux, 2018-10-14:
@dan9thsense I was able to fix my problem by [installing](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) RealSense SDK 2.0+ from source.  Note that I'm running 16.04 with ROS Kinetic.  Note that I also had to build and apply the kernel patch. Aside from that, you might want to make sure the firmware on the camera is supported by whatever version RealSense SDK you are installing. If not, you should [update the firmware](https://www.intel.com/content/www/us/en/support/articles/000028171/emerging-technologies/intel-realsense-technology.html). Good luck!

dan9thsense, 2018-10-16:
@jmichaux thanks for the suggestions.  I will try them.

closed 2019-04-04 by RealSenseCustomerSupport