realsenseai/realsense-ros #441: Install from source vs through Debian pkg

Achllle, 2018-08-15:
I installed librealsense2 via Debian package, but when I `catkin_make` I get the message that the SDK is missing:

```
Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases
```

`librealsense` provides additional [installation instructions](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md) for Ubuntu but it seems like these require it to be installed from source.

How can I install using the debian package and if not, can the readme be clarified or updated?

`dpkg -L librealsense2` yields

```
/.
/usr
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.15.0
/usr/share
/usr/share/doc
/usr/share/doc/librealsense2
/usr/share/doc/librealsense2/changelog.Debian.gz
/usr/share/doc/librealsense2/copyright
/usr/lib/x86_64-linux-gnu/librealsense2.so.2
```

doronhi, 2018-08-22:
It seems to me that you did: 
sudo apt-get install librealsense2

instead of following the instructions linked under  "Install from Debian Package" in the **intel-ros** installation instructions page: https://github.com/intel-ros/realsense
This link sends you to the page you describedq as "additional installation instructions" which is the instructions to install the necessary **librealsense2** packages. 
The package you installed, librealsense2, contains the .so file and not the headers needed for the build of the realsense2_camera ROS package.
I hoped it clarifies things a bit.

Achllle, 2018-08-22:
Thanks for your reply. I did install through [Install from Debian package](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages). 
So if I understand correctly, the debian package listed above is NOT compatible with the realsense2_camera ROS package? In that case, why is that installation listed as an option?

doronhi, 2018-09-03:
I'm sorry, I don't seem to follow you:
There are 2 options for installing librealsense SDK 2.0 listed in the page: https://github.com/intel-ros/realsense: 
"Install from Debian Package", 
"Build from sources by downloading...".
Both seems to work fine for the purpose of building realsense2_camera ROS package.
The option of "apt-get install librealsense2" is not listed in the building instructions for the ROS package.
What option do you refer to in the last sentence: "why is that installation listed as an option"?
I'm sorry for getting back to you after such a while. I do try to understand the problem.

Achllle, 2018-09-05:
Again, I did not install through `apt-get install librealsense2`, but I followed the Debian instructions. And that simply does not work for me, yielding the error above. I.e. your statement "Both seems to work fine for the purpose of building realsense2_camera ROS package." does not hold true for me.

The reason I said 'why is that installation listed as an option' was because I understood from your answer that installing through Debian package was not compatible with the realsense2_camera ROS package.

doronhi, 2018-09-06:
So, If I understand you correctly, the main issue now is that building the ROS package realsense2_camera fails when installing librealsense using the instructions set under "Install from Debian Package" (https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages)
Right?

It's basically the same set of instructions listed in the .travis.yml file in the repository, used to install and check the package on Travis so something is different.
Can you give me more details, like what OS you are using? Did you try that on a clean machine? Can you add the full history of commands or the full log so that I'll try to recreate it?

I hope that when that problem is solved it'll be easier to address the documentation issue.

vatbrain, 2018-09-09:
I just went through these instructions again today to re-install the realsense SDK and encountered the 
"Intel RealSense SDK 2.0 is missing" when I ran "catkin_make clean" before attempting my catkin_make.
I find the instructions pretty confusing.
What was not clear to me was that it is necessary to also install the additional packages librealsense2-dev and librealsense2-dbg in order to use the library with this package. Installing those packages solved the "Intel RealSense SDK 2.0 is missing" problem for me.

doronhi, 2018-09-12:
librealsense2-dev is, indeed, needed for the building of realsense2_camera but not librealsense2-dbg.
I agree that the instructions are not clear enough. I will try to get around to it in the near future. 
In the meanwhile and until a more robust installation tool is available, you can check out the .travis.yaml file in the root directory. It contains a script for building and installing the package from scratch.

Achllle, 2018-09-13:
Thanks for clarifying, @doronhi 

I removed my source installation and tried again. This time around it seems to work (no idea what was different). Clearer instructions would be useful. Closing for now.

closed 2018-09-13 by Achllle

Achllle, 2018-09-21:
@doronhi just to be absolutely clear: you realize that the librealsense2-dev package is under the section 'Developers shall install additional packages:', making it unclear that it's a hard requirement for the realsense ROS package?
So that was what was different in my last comment!

doronhi, 2018-09-26:
Thank you for clarifying.
I updated the README.md. I would appreciate you opinion.

drzraf, 2018-11-02:
AFAICT the current README makes clear that ROS-part needs to be built.
Personally I need only ros  to decode/export a video sequence grabbed from rs-viewer (a workaround for https://github.com/IntelRealSense/librealsense/issues/2619#issuecomment-433459861).
Having a debian package instead of building this would be nice.