realsenseai/realsense-ros #442: API version mismatch!

Serjey2017, 2018-08-18:
Hello everybody! Few month ego i was successfully compiled realsense d435 sdk for linux and all my programs worked fine but now when i try compile my code again i got fault and its says:

RealSense error calling rs2_create_context(api_version:21400):
    API version mismatch: librealsense.so was compiled with API version 2.15.0 but the application was compiled with 2.14.0! Make sure correct version of the library is installed (make install)
Press <RETURN> to close this window...

I am using Linux Mint 19 with 4.15.0-32-generic kernel, but when i compile sdk at every first time there was 4.15.0-29 kernel version. Also use QTcreator with gcc version 7.3.0 (Ubuntu 7.3.0-16ubuntu3).

By the way native realsense viewer stay works fine! Where is the problem?

freemanlo, 2018-08-19:
Hi @Serjey2017,

You seem to have librealsense 2.15.0 already installed, most likely as a result of sudo make install. If you have previously built the library from source, navigate to build folder and run sudo make uninstall. Alternatively, you can look under /usr/local/bin, /usr/local/lib and /usr/local/include and remove any remains of version 2.15.0. 
Make sure to compile your code and run your app with the matched librealsense version.

Serjey2017, 2018-08-19:
Yes i have done that: in source directory run make uninstall and for now all works fine. Thank you!

closed 2018-08-19 by Serjey2017