realsenseai/realsense-ros #443: frames didn't arrived within 5 seconds

qiuchenli, 2018-08-20:
Dear sir,:
I use realsense d435 in ubuntu 16.04 for the use of pointcloud,when use
$ roslaunch realsense2_camera rs_rgbd.in terminal, 
sometims it didn't get pointcloud infomation and the terminal get the info that "frames didn't arrived within 5 seconds",any solutions?
PS: I tried the solutions that add "reset device",but found that would affect the fps,(almost 7fps).

fiorano10, 2018-08-28:
@qiuchenli This has been documented here:[Frames didn't arrive error](https://github.com/IntelRealSense/librealsense/issues/1086)
If it's the PointCloud then the frame rate would remain stable ~6-7FPS even before the reset.

The pcl library depends heavily on the clock speed of the CPU. Do you get a higher frame rate before resetting the camera? If so, try unplugging and replugging the camera, and see if that affects the frame rate as well.

qiuchenli, 2018-09-05:
@fiorano10 I tried the reset ros wrapper(https://github.com/icarpis/realsense/tree/reset_dev) in my laptop and the FPS down from 30Hz to 26Hz, but in the machine, the FPS down from 27Hz to 16Hz, I think maybe the problem is my hardware, the reset policy has no problem. Thanks for a lot!

fiorano10, 2018-09-05:
@qiuchenli ,please close the issues that have been resolved. Thanks!

closed 2018-09-10 by qiuchenli