realsenseai/realsense-ros #445: Error running RealSense D435 on Ubuntu 16.04(kernel 4.4.0-131-generic)

arya001, 2018-08-21:
I am getting an error message when running `realsense-viewer`. The realsense window does appear and I can also run the launch files.

    WARNING [139658086876928] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]

fiorano10, 2018-08-27:
It's a warning message, does it affect your image stream in any way? Can you view the image_stream? I get the same warning, but it hasn't impacted the functionality in any way, yet.

arya001, 2018-08-28:
@fiorano10 No, the image stream is alright. But I always get that WARNING on terminal when I run any realsense file. 

Another problem I am facing is having to hardware reset everytime I run a launchfile.

fiorano10, 2018-08-28:
The hardware reset issue has been logged multiple times, there is no fix yet, even with the new FW, or SDK updates. I've been using the ResetDevice node to reset the RealSense before re-running a RealSense launch file, or I keep it running. [Frames didn't arrive error](https://github.com/IntelRealSense/librealsense/issues/1086)

arya001, 2018-08-28:
@fiorano10 Alright. Thanks for the help

lchop, 2018-09-07:
Hello,
@fiorano10, where can I find thid ResetDevice node ?
Best regards,

fiorano10, 2018-09-07:
@lchop Here you go [reset_device](https://github.com/fiorano10/realsense/blob/devel/realsense2_camera/src/reset_device.cpp)

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

fiorano10, 2019-03-07:
@arya001 Please close the issue, if your problem was solved.

closed 2019-04-05 by arya001