realsenseai/realsense-ros #451: Does not build against librealsense v2.16.0

alexswerner, 2018-08-27:
I tried to build this and librealsense as ROS packages with catkin on Ubuntu xenial and ROS kinetic. Against v2.15.0 it builds successfully but against v2.16.0 it fails. The error is:
<pre>
In file included from /home/workspace/devel/include/librealsense2/hpp/rs_device.hpp:8:0,
                 from /home/workspace/devel/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /home/workspace/devel/include/librealsense2/hpp/rs_context.hpp:8,
                 from /home/workspace/devel/include/librealsense2/rs.hpp:9,
                 from /home/workspace/src/realsense/realsense2_camera/src/../include/../include/realsense_node_factory.h:11,
                 from /home/workspace/src/realsense/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/workspace/src/realsense/realsense2_camera/src/base_realsense_node.cpp:1:
/home/workspace/devel/include/librealsense2/hpp/rs_sensor.hpp: In instantiation of 'void rs2::frame_callback<T>::on_frame(rs2_frame*) [with T = rs2::asynchronous_syncer; rs2_frame = rs2_frame]':
/home/workspace/src/realsense/realsense2_camera/src/base_realsense_node.cpp:1396:1:   required from here
/home/workspace/devel/include/librealsense2/hpp/rs_sensor.hpp:108:13: error: no match for call to '(rs2::asynchronous_syncer) (rs2::frame)'
             on_frame_function(frame{ fref });
             ^
</pre>

When i run git bisect the first offending commit is 6850677e0473bd01a2078ef6e0398a617539cf56 however
with a different error message:
<pre>
/home/workspace/src/librealsense/src/../include/librealsense2/hpp/rs_frame.hpp: In instantiation of 'void rs2::frameset::foreach(T) const [w
ith T = librealsense::align::should_process(const rs2::frame&)::<lambda(rs2::frame&)>]':
/home/workspace/src/librealsense/src/proc/align.cpp:572:134:   required from here
/home/workspace/src/librealsense/src/../include/librealsense2/hpp/rs_frame.hpp:965:23: error: no match for call to '(librealsense::align::sh
ould_process(const rs2::frame&)::<lambda(rs2::frame&)>) (rs2::frame)'
                 action(frame(fref));
</pre>

I was successful locating the original api change which causes the build to fail. Staying with librealsense v2.15.0 is not an option as it is not compatible with kernels >=4.16.

Let me know if I can provide more information.

atyshka, 2018-08-27:
Also having this error

Jaeyoung-Lim, 2018-08-28:
Same problem.

Downgrading the librealsense to 2.14 helped

ChainBreak, 2018-08-29:
Same problem. Making me sad

robotlinker, 2018-08-29:
I have the same problem
[image]

robotlinker, 2018-08-29:
same problem with latest SKD

doronhi, 2018-08-29:
Fixed. Checkout latest release, Pull request #452.

nindanaoto, 2018-08-29:
The build is now passing with latest commit against librealsense v2.16.0.
Thank you.

itabr, 2018-08-29:
I am still getting the same problem with latest Intel® RealSense™ SDK 2.0. I tried building from source and got the same error.

bhaskara, 2018-08-29:
I get a compile error about `invoke` not being found when I try to build this.  Suggestions?

nindanaoto, 2018-08-30:
I used the librealsense provided by ubuntu repository for 18.04, not the one builded from the source.

bhaskara, 2018-08-30:
I'm on 16.04 + ROS Kinetic, using the ubuntu librealsense2 2.15.  Is this a supported configuration @doronhi ?

doronhi, 2018-08-30:
@bhaskara, The latest update on intel-ros is checked with librealsense2 2.**16**. The same configuration is as you said.

doronhi, 2018-08-30:
@itabr , Is it possible that you still have remnants of previous versions on your machine? Can you confirm that you get the same error on a clean machine, using the latest commit of intel-ros and librealsense SDK 2.0 (build 2.16.0)? If so, can you provide additional data, such as a build log, what version of OS you are using, etc.?

bhaskara, 2018-08-30:
Thanks @doronhi .  My specific error was that it couldn't find the function `invoke` from your patch.  Do you know where that's from?  Is it a librealsense thing?

itabr, 2018-08-30:
I can confirm that intel-ros/realsense with the latest master commit works with IntelRealSense/librealsense v2.16.0 (Installed from [Debian)](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) and Ubuntu 16.04.

doronhi, 2018-08-31:
@bhaskara, invoke is indeed a librealsense thing...
It belongs to class asynchronous_syncer : public processing_block (defined in file: include\librealsense2\hpp\rs_processing.hpp)

bhaskara, 2018-08-31:
Ah, thanks for the pointer.  Looking at the source, in 2.15, `asynchronous_syncer` does *not* inherit from `processing_block`, and thus `PipelineSyncer` does not inherit `invoke` (https://github.com/IntelRealSense/librealsense/blob/1e8aa8002bc3d6afbfc7b733ea4a19864bbeca14/include/librealsense2/hpp/rs_processing.hpp#L330v).  It looks, therefore, like upgrading to the above patch requires simultaneously upgrading to 2.16.  

It might be convenient to have a version of the patch that allowed 2.15 (e.g., in my setup the choice of librealsense2 version and the choice of intel-ros version happen in two different parts of the provisioning and would be nice to integration test independently).    I tried just copying over the definition of invoke into `base_realsense_node.h`, but got further errors (about not finding the `error` class) that I haven't resolved yet.

Let me know if any of this seems way off base.

yeyanlei, 2018-09-05:
[image]
@doronhi , I meet the same problem( error: no match for call to ‘(rs2::asynchronous_syncer) (rs2::frame)’). could you give me some suggestion. Thanks for your help

Achllle, 2018-09-06:
#452 made building fail with librealsense 2.15 indeed.

AndyZe, 2018-09-28:
I am getting the invoke() error as well. I installed librealsense by binary (according to this page: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages). The git commit of the realsense package I cloned is 6f2c7c8