realsenseai/realsense-ros #454: Unable to turn off Depth Emitter by default

fiorano10, 2018-08-29:
I'm on ROS-Kinetic/ Xenial, RealSense FW 5_10_03, librealsense2 V2.15.0. Even if I set the depth emitter to 0 in the cfg file, it still continues to work. The only way to shut it off is to have a bash script run the dynamic_reconfigure command to set it to 0.

Rayman, 2018-10-19:
I fixed this issue in this PR: https://github.com/intel-ros/realsense/pull/481. Can you try it out?

doronhi, 2018-10-21:
I tried PR #481. It didn't solve the problem, as I understand it. Admittedly, I didn't understand exactly what it does and I will be very happy with a more elaborated explanation since I don't consider myself a ROS expert.
However, I did find a problem in the initialization process of the reconfigurable parameters and fixed it in PR #499. @fiorano10 , Could you confirm that any of the PRs suggested here fix your problem?

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

closed 2019-03-07 by fiorano10

fiorano10, 2019-08-13:
@doronhi The fix did work