realsenseai/realsense-ros #459: d435 urdf displays mesh incorrectly

rongoldman-tri, 2018-09-06:
Hi. We just added the D435 urdf to our robot model and when it is displayed the camera orientation is off by 90 degrees.

[image]

This is showing the camera_link frame and also the camera_right_ir_optical_frame.

By changing line 49 so the roll is `${M_PI}` instead of `${M_PI/2}`  to:

      `<origin xyz="${d435_cam_mount_from_center_offset} 0.0 ${d435_cam_height/2}" rpy="${M_PI} 0 ${M_PI/2}"/>`

it displays properly:

[image]

Note the origin tag doesn't seem to affect displaying a box, only the mesh has the problem. Whether roll is pi or pi/2 the display for the box is:

[image]

thanks,

-- Ron --

doronhi, 2018-09-12:
Hi @rongoldman-tri,
I can't recreate the situation you presented. I run the following command:
roslaunch realsense2_camera view_d435_model.launch

I then Show camera_right_ir_optical_frame and camera_link, set Market Scale to 0.1 and get the following image:
[image]

Can you tell what are we doing differently?

rongoldman-tri, 2018-09-12:
Thanks for looking into this. The problem was on our side: we had converted the d435.dae mesh to use STL format. The converted STL file introduced the rotation I reported as a bug. Redoing the conversion with MeshLab created a good STL file that preserved the original mesh orientation. So everything is fine & I am closing the issue. Sorry to have bothered you.

-- Ron --

closed 2018-09-12 by rongoldman-tri

doronhi, 2018-09-13:
No problem. Glad it worked out for you.