realsenseai/realsense-ros #460: how to get 32FC1 from /camera/depth/image_rect_raw

gongyue666, 2018-09-06:
hi, I want to get sensor_msgs/Imagic from  /camera/depth/image_rect_raw with 32FC1, I have changed the rs_rgbd.launch, set align_depth = false. There are some new topics of sw_registered, but on them the image_encodings is still 16UC1.So where is my fault?

dimaw24, 2019-02-06:
have you found a solution?

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@gongyue666 What's the realsense ROS version when you get the issue? What camera and FW version? Did you still get it with the latest version? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

@gongyue666 Could you please provide more information so that we can further investigate? Thanks!

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@gongyue666 Haven't got update from you for long time. If no other questions, will close the ticket accordingly.

closed 2019-05-03 by RealSenseCustomerSupport

ShirleyZYJ, 2019-06-27:
@RealSenseCustomerSupport 

Hi,
I am facing the same problem. I am using Ubuntu 18.04 and ROS Melodic. The realsense camera I used is D415. The image published by "/camera/depth/image_rect_raw" topic launching by rs_rgbd.launch is in 16UC1, but what I need is 32FC1 format.
Any suggestion on the format conversion?
Thanks

doronhi, 2019-06-30:
Each pixel is depth in millimeters.

SuperKuooo, 2019-10-15:
@doronhi Thanks for mentioning the pixel is in millimeters. I might have missed it, but could you put this information somewhere more obvious? It took me a little digging to find this. Thanks!

doronhi, 2019-10-15:
Thanks for the remark.
realsense2_camera follows OpenNI Raw Representation as the ROS convention mentioned [here](https://www.ros.org/reps/rep-0118.html).
However, I accept that it is not clear for the realsense user and will add it to the README.md file.