realsenseai/realsense-ros #463: Set shutter speed (not fps) for video recording?

abarajithan11, 2018-09-10:
Hi,

I've attached three Realsense D435 cameras on a wheeled robot and I'm recording videos while driving it around. I'm controlling the camera through ROS (Intel relasense2_camera package) This way, I'm getting a video with each frame blurred. Is there a way to set the shutter speed?

Thanks

abarajithan11, 2018-09-12:
I'm moving at around 0.5 m/s and I'm looking at RGB stream. I need to train a DNN using the RGB frames. Isn't there an option to use a faster shutter speed?

fiorano10, 2018-09-12:
I don’t think you can change the shutter speed. The RGB camera is a rolling shutter which causes motion blur, and I don’t know if there’s a way around that. You’re best bet would be to use the infrared cameras instead. 

If you’re training the DNN for object detection then the RGB image stream isn’t required. I might be able to assist better if you could tell me what the task is.

stefangordon, 2018-11-22:
Rolling shutter is going to cause distortion, not blur.   Regardless of the shutter type, each individual picture has an exposure time which is the cause of the blur.

doronhi, 2019-01-28:
You can indeed change the exposure time using rqt_reconfigure. The image will be sharper but also darker. You'll may need a scene with more light.
In addition to what @stefangordon said, rolling shutter will cause different parts of the image to be taken in different times (the effect of propeller where you can the same blade in several locations...) but this effect is only visible for very fast moving objects.
If your problem is indeed concerned with a fast moving scene and rgb image is necessary you can try the D415 with global shutter rgb camera.

closed 2019-01-28 by doronhi

blubbi321, 2019-02-22:
@doronhi I also have the problem of a rapidly moving scene and the RGB stream being required. However, as far as I can see there is no D415 with global shutter for RGB and the D435 only has global shutter sensors on the IR streams.
In case this is not right could you tell be where to get a D415 with a global shutter RGB sensor?

doronhi, 2019-02-24:
I see that I got mixed up before.
The cameras have the following sensors:
D415 – depth - rolling shutter. RGB – rolling shutter.
D435 – depth – Global shutter. RGB – rolling shutter.

There is no D415 with global shutter on the RGB sensor.
Sorry about the misinformation.

blubbi321, 2019-02-24:
Ok, too bad. Thanks for the clarification!