realsenseai/realsense-ros #464: Issue launching the camera

FYPBois, 2018-09-10:
Hi,

We are having an issue running the realsense camera through ROS. Having gotten catkin_make to run without errors. However, getting an error (shown below) when trying to launch with the following command. Please let me know if anyone has any ideas to get this issue fixed.

roslaunch realsense2_camera rs_camera.launch

Error:

[ INFO] [1536546180.227283302]: Initializing nodelet with 4 worker threads.
[ERROR] [1536546180.329636650]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1536546180.329744181]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1536546180.330428662]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-2] process has died [pid 9247, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/fypbois/.ros/log/65e9f26c-b49e-11e8-bb44-e442a68f1f6f/camera-realsense2_camera-2.log].

SteveMacenski, 2018-09-14:
Did you source your workspace?

FYPBois, 2018-09-18:
It is there when i run rospack find.

RealSenseCustomerSupport, 2019-04-01:
----------------------------------------------

Close this issue for lack of activity. Please feel free to reopen if the issue is still relevant.

closed 2019-04-09 by RealSenseCustomerSupport

nikkolasedi, 2020-01-28:
Do you have the solution?

hwb0314, 2020-03-04:
I have the exact same problem. Do you have the solution?