realsenseai/realsense-ros #465: No RGB data on Odroid XU4

Seiwert, 2018-09-10:
Required Info |  
-- | --
Camera Model | D415
Firmware Version | 05.09.14.00
Operating System & Version | Ubuntu 16 minimal
Kernel Version (Linux Only) | Linux  4.14.55
Platform | odroid XU4
SDK Version | build from source librealsense v2.16.0
Language | C
Segment | others

I launch the camera normally as such:
``
roslaunch realsense2_camera rs_camera.launch
``

After doing so, I see this output:
```
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [1443]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a7c3d09e-b4b8-11e8-9174-0013ef200c4d
process[rosout-1]: started with pid [1456]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [1473]
process[camera/realsense2_camera-3]: started with pid [1474]
[ INFO] [1536556583.657406849]: Initializing nodelet with 8 worker threads.
[ INFO] [1536556583.787768272]: RealSense ROS v2.0.3
[ INFO] [1536556583.787905777]: Running with LibRealSense v2.16.0
[ INFO] [1536556584.178908944]: getParameters...
[ INFO] [1536556584.321949089]: setupDevice...
[ INFO] [1536556584.322113553]: JSON file is not provided
[ INFO] [1536556584.322271308]: ROS Node Namespace: camera
[ INFO] [1536556584.322455981]: Device Name: Intel RealSense D415
[ INFO] [1536556584.322623195]: Device Serial No: 816612060248
[ INFO] [1536556584.322816285]: Device FW version: 05.09.02.00
[ INFO] [1536556584.322979916]: Device Product ID: 0x0AD3
[ INFO] [1536556584.323286384]: Enable PointCloud: Off
[ INFO] [1536556584.323426472]: Align Depth: Off
[ INFO] [1536556584.323584311]: Sync Mode: Off
[ INFO] [1536556584.324726725]: Device Sensors: 
[ INFO] [1536556584.325433749]: Stereo Module was found.
[ INFO] [1536556584.325636797]: RGB Camera was found.
[ INFO] [1536556584.326384323]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1536556584.326524494]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1536556584.326664707]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1536556584.326904132]: setupPublishers...
[ INFO] [1536556584.337651330]: Expected frequency for depth = 30.00000
[ INFO] [1536556584.351088494]: Expected frequency for infra1 = 30.00000
[ INFO] [1536556584.357904517]: Expected frequency for infra2 = 30.00000
[ INFO] [1536556584.364526742]: Expected frequency for color = 30.00000
[ INFO] [1536556584.377013957]: setupStreams...
[ INFO] [1536556584.428533789]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1536556584.429949753]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1536556584.431236714]: infra2 stream is enabled - width: 640, height: 480, fps: 30
 10/09 05:16:24,437 WARNING [3069820928] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 10/09 05:16:24,439 WARNING [3069820928] (sensor.cpp:338) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1536556584.487727255]: color stream is enabled - width: 640, height: 480, fps: 30
 10/09 05:16:24,522 ERROR [2833249312] (backend-v4l2.cpp:1002) Video frame arrived in idle mode.
 10/09 05:16:24,522 ERROR [2833249312] (backend-v4l2.cpp:1002) Video frame arrived in idle mode.
 10/09 05:16:24,523 ERROR [2812277792] (backend-v4l2.cpp:1002) Video frame arrived in idle mode.
[ INFO] [1536556584.550960859]: publishStaticTransforms...
[ INFO] [1536556584.553588282]: RealSense Node Is Up!
```
After doing this, I can echo depth topics fine and render point cloud information in RVIZ fine. However, I am unable to retrieve any color data. Echoing the **/camera/color/image_raw** topic just returns zeros. I believe the issue is related to this [other issue in the librealsense repository](https://github.com/IntelRealSense/librealsense/issues/2306) but some confirmation would be nice. I haven't downgraded to 2.15 to verify yet.

Thanks in advance!

SteveMacenski, 2018-09-14:
>  05.09.02.00

dorodnic, 2018-09-27:
Hi @Seiwert 
Please try todays release (it works with 2.16.1 LRS which solves this issue).

Seiwert, 2018-11-12:
Hey @dorodnic 

Sorry for the delay. Was just able to confirm that 2.16.1 fixes this issue. Thanks again for the help!