realsenseai/realsense-ros #467: Create a release with static librealsense versions

SteveMacenski, 2018-09-11:
It's been several months with several new librealsense versions brought out and its never clear what will work with what. Can we get back to releasing a static release that works with a given librealsense version?

https://github.com/intel-ros/realsense/releases

SteveMacenski, 2018-09-11:
This is super important for those using these cameras in a production environment. If the master branch stops working with a version or a bug is introduced, production resources can be brought down too. There hasn't been a release since March with matching supported librealsense SDK and firmware versions.

doronhi, 2018-09-12:
I totally agree and aware of the issue. Working on it and plan to freeze a version in the near future.

SteveMacenski, 2018-09-12:
Awesome, thanks for all your hard work here!

closed 2018-09-12 by SteveMacenski