realsenseai/realsense-ros #469: What's the serial_no for d435 camera should I input in the launch file?

1075976432, 2018-09-17:
It seems that the TF frames were not same with the real d435 camera model after launched the file "rs_camera.launch". 
I guess the reason was miss of the serial_no in the launch file, so the default model was not d435 but other type.
Dose anyone know the serial_no for d435?
Thanks!

neomanic, 2018-09-20:
Run `rs-enumerate-devices | grep Serial` to get the serial number.

1075976432, 2018-09-20:
@neomanic 
Thanks at first!
But the the result was same after input the serial_no gained by your method. 
The TF between **camera_link** and **camera_color_optical_link** was still not correct with the urdf model loaded by the view_d435.launch file which I think has the true frames parameters.
Maybe there is no option to change the TF frames in this Ros package when launching rs_camera.launch.

neomanic, 2018-09-21:
Ah, okay. You're saying the tfs provided by the URDF doesn't match reality? I can't help much then, I've been happy with them, but admittedly I haven't looked too closely at them.

fiorano10, 2018-09-25:
@1075976432 I think the urdf has been updated in the new release, which should make more sense.

doronhi, 2018-10-11:
@1075976432, Did the update fix it?

1075976432, 2018-10-26:
@doronhi 
The urdf has the correct parameters of frames. However, when launched the **rs_camera.launch** file, the problem still exist.
I solve the problem by broadcasting a right frame....

doronhi, 2018-10-26:
I am not sure what you mean by “broadcasting a right frame “. I am not sure what is the exact problem you describe.
Is your issue related to #497?

atyshka, 2018-11-13:
Here is a detailed video of the issue. I started the model which had the correct transforms but then everything is offset when the driver is started.
[image]
When the driver is started the frames no longer line up with the geometry.

PR #567 (2019-01-23): Fixed different transforms between xacro and driver
This fixes #469 and #497 

The d435 URDF had an error where there was a non-zero transform for camera_link but the driver output a zero origin for this link. This caused the transforms to shift between starting a robot description and starting the driver. This commit brings the transforms in line with the driver while keeping the visual and collision geometry the same.