realsenseai/realsense-ros #470: roslaunch realsense2_camera rs_rgbd.launch   FAILED!

jessieqiqi, 2018-09-18:
When  I roslaunch realsense2_camera rs_rgbd.launch , but it shows below . What's the reason? I didn't change any parameters in the launch file. My device is D435.

haharobot@haharobot-minzy:~$ roslaunch realsense2_camera rs_rgbd.launch
... logging to /home/hitrobot/.ros/log/d9d2df54-bb06-11e8-a5c5-f0d5bf32678e/roslaunch-hitrobot-minzy-1014.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hitrobot-minzy:40909/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /camera/
    color_rectify_color (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [1026]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d9d2df54-bb06-11e8-a5c5-f0d5bf32678e
process[rosout-1]: started with pid [1040]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [1057]
process[camera/realsense2_camera-3]: started with pid [1058]
process[camera/color_rectify_color-4]: started with pid [1059]
process[camera/points_xyzrgb_hw_registered-5]: started with pid [1069]
[ INFO] [1537249874.112857537]: Initializing nodelet with 4 worker threads.
[ INFO] [1537249874.252087373]: RealSense ROS v2.0.3
[ INFO] [1537249874.252122624]: Running with LibRealSense v2.16.0
[ INFO] [1537249874.529246011]: getParameters...
[ INFO] [1537249874.579393574]: setupDevice...
[ INFO] [1537249874.579427211]: JSON file is not provided
[ INFO] [1537249874.579450652]: ROS Node Namespace: camera
[ INFO] [1537249874.579478916]: Device Name: Intel RealSense USB2
[ INFO] [1537249874.579494759]: Device Serial No: 752112070181
[ INFO] [1537249874.579512531]: Device FW version: 05.08.15.00
[ INFO] [1537249874.579530331]: Device Product ID: 0x0AD6
[ INFO] [1537249874.579547931]: Enable PointCloud: Off
[ INFO] [1537249874.579563170]: Align Depth: On
[ INFO] [1537249874.579574193]: Sync Mode: On
[ INFO] [1537249874.580052123]: Device Sensors: 
[ INFO] [1537249874.580321217]: Stereo Module was found.
[ INFO] [1537249874.580606895]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1537249874.580623394]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1537249874.580637890]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1537249874.580652569]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1537249874.580713199]: setupPublishers...
[ INFO] [1537249874.584745883]: Expected frequency for depth = 30.00000
[ INFO] [1537249874.620448562]: Expected frequency for infra1 = 30.00000
[ INFO] [1537249874.655182513]: Expected frequency for aligned_depth_to_infra1 = 30.00000
[ INFO] [1537249874.700269824]: Expected frequency for infra2 = 30.00000
[ INFO] [1537249874.773487329]: Expected frequency for aligned_depth_to_infra2 = 30.00000
[ INFO] [1537249874.841426409]: setupStreams...
 18/09 13:51:14,846 WARNING [140018650625984] (sensor.cpp:338) Unregistered Media formats : [ UYVY Y8   ]; Supported: [ ]
[ WARN] [1537249874.849296738]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Format: Z16, Width: 640, Height: 480, FPS: 30
[ WARN] [1537249874.849442734]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 1, Format: Y8, Width: 640, Height: 480, FPS: 30
[ WARN] [1537249874.849531409]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Format: Y8, Width: 640, Height: 480, FPS: 30
[ INFO] [1537249874.849918667]: publishStaticTransforms...
[ERROR] [1537249874.850206122]: Given depth profile is not supported by current device!
[FATAL] [1537249875.161035040]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[FATAL] [1537249875.162339029]: Failed to load nodelet '/camera/points_xyzrgb_hw_registered` of type `depth_image_proc/point_cloud_xyzrgb` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-2] process has died [pid 1057, exit code 1, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/hitrobot/.ros/log/d9d2df54-bb06-11e8-a5c5-f0d5bf32678e/camera-realsense2_camera_manager-2.log].
log file: /home/hitrobot/.ros/log/d9d2df54-bb06-11e8-a5c5-f0d5bf32678e/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has died [pid 1058, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/hitrobot/.ros/log/d9d2df54-bb06-11e8-a5c5-f0d5bf32678e/camera-realsense2_camera-3.log].
log file: /home/hitrobot/.ros/log/d9d2df54-bb06-11e8-a5c5-f0d5bf32678e/camera-realsense2_camera-3*.log
[camera/points_xyzrgb_hw_registered-5] process has died [pid 1069, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb realsense2_camera_manager false rgb/image_rect_color:=color/image_rect_color rgb/camera_info:=color/camera_info depth_registered/image_rect:=aligned_depth_to_color/image_raw depth_registered/points:=depth_registered/points __name:=points_xyzrgb_hw_registered __log:=/home/hitrobot/.ros/log/d9d2df54-bb06-11e8-a5c5-f0d5bf32678e/camera-points_xyzrgb_hw_registered-5.log].
log file: /home/hitrobot/.ros/log/d9d2df54-bb06-11e8-a5c5-f0d5bf32678e/camera-points_xyzrgb_hw_registered-5*.log

neomanic, 2018-09-20:
This looks to me like you've connected via USB2, not USB3. Run `lsusb` to get which bus/device is the Realsense (it'll be vendor:id 8086:0b07), then run `lsusb -t` and make sure it says 5000M on the end (and not 480M).

manhha1402, 2018-12-19:
Hi,
I had the same problem when launching rs_rgbd.launch with high resolution size (1920x1080). The topic "/camera/aligned_depth_to_color/image_raw" has no data. It seems like depth maps are unable to upsample from VGA resolution to 1080p resolution. Any one has the same problem ?
Thanks
[image]
[image]

RealSenseCustomerSupport, 2019-04-03:
----------------------------------------------

@jessieqiqi,

Is your issue resolved?

@manhha1402,
Any test result based on the new librealsense and ros wrapper?
Per the screenshot, it looks SR300 is used. Depth stream is set to VGA. Will reducing RGB resolution from 1080p to VGA have any problem?

manhha1402, 2019-04-03:
Hi, 
The problem is solved. Thanks

closed 2019-05-13 by RealSenseCustomerSupport