realsenseai/realsense-ros #480: d415 pointcloud is noisy in rviz

hi-kondo, 2018-09-25:
Hi, I'm using d415.
```
firmware:5_10_3_0
librealsense: Intel® RealSense™ SDK 2.0 (build 2.16.0)
```
Below warning shows,
```
$ roslaunch realsense2_camera rs_rgbd.launch 
... logging to /home/seed/.ros/log/cc11cb64-c0a1-11e8-84cb-a434d993f1c0/roslaunch-seed-ThinkPad-T460s-31679.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://seed-ThinkPad-T460s:41421/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    color_rectify_color (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [31689]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to cc11cb64-c0a1-11e8-84cb-a434d993f1c0
process[rosout-1]: started with pid [31703]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [31720]
process[camera/realsense2_camera-3]: started with pid [31721]
process[camera/color_rectify_color-4]: started with pid [31722]
process[camera/points_xyzrgb_hw_registered-5]: started with pid [31730]
[ INFO] [1537866178.895956981]: Initializing nodelet with 4 worker threads.
[ INFO] [1537866178.994602165]: RealSense ROS v2.0.3
[ INFO] [1537866178.994634329]: Running with LibRealSense v2.16.0
[ INFO] [1537866179.297975607]: getParameters...
[ INFO] [1537866179.358371948]: setupDevice...
[ INFO] [1537866179.358442328]: JSON file is not provided
[ INFO] [1537866179.358474861]: ROS Node Namespace: camera
[ INFO] [1537866179.358513235]: Device Name: Intel RealSense D415
[ INFO] [1537866179.358534864]: Device Serial No: 816312060493
[ INFO] [1537866179.358559419]: Device FW version: 05.10.03.00
[ INFO] [1537866179.358582056]: Device Product ID: 0x0AD3
[ INFO] [1537866179.358609188]: Enable PointCloud: Off
[ INFO] [1537866179.358637472]: Align Depth: On
[ INFO] [1537866179.358658196]: Sync Mode: On
[ INFO] [1537866179.358970824]: Device Sensors: 
[ INFO] [1537866179.359174501]: Stereo Module was found.
[ INFO] [1537866179.359223837]: RGB Camera was found.
[ INFO] [1537866179.359450585]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1537866179.359481659]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1537866179.359509306]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1537866179.359564560]: setupPublishers...
[ INFO] [1537866179.363307788]: Expected frequency for depth = 30.00000
[ INFO] [1537866179.395862362]: Expected frequency for infra1 = 30.00000
[ INFO] [1537866179.421994418]: Expected frequency for aligned_depth_to_infra1 = 30.00000
[ INFO] [1537866179.446018089]: Expected frequency for infra2 = 30.00000
[ INFO] [1537866179.471695361]: Expected frequency for aligned_depth_to_infra2 = 30.00000
[ INFO] [1537866179.496306551]: Expected frequency for color = 30.00000
[ INFO] [1537866179.522022056]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1537866179.551641932]: setupStreams...
[ INFO] [1537866179.561624840]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1537866179.562019535]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1537866179.562349605]: infra2 stream is enabled - width: 640, height: 480, fps: 30
 25/09 18:02:59,564 WARNING [140294086170496] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 25/09 18:02:59,565 WARNING [140294086170496] (sensor.cpp:338) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1537866179.573798384]: color stream is enabled - width: 640, height: 480, fps: 30
 25/09 18:02:59,589 ERROR [140293388445440] (backend-v4l2.cpp:1002) Video frame arrived in idle mode.
 25/09 18:02:59,589 ERROR [140293388445440] (backend-v4l2.cpp:1002) Video frame arrived in idle mode.
[ INFO] [1537866179.597572517]: publishStaticTransforms...
[ INFO] [1537866179.598323409]: RealSense Node Is Up!
 25/09 18:02:59,786 WARNING [140293158708992] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
 25/09 18:02:59,807 WARNING [140293371086592] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1537866179.817030484]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 25/09 18:03:04,674 WARNING [140294086170496] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error
[ WARN] [1537866184.675410338]: Reconfigure callback failed with exception get_xu(id=11) failed! Last Error: Input/output error: 
[ WARN] [1537866370.065060462]: Messages of type 2 arrived out of order (will print only once)
[ WARN] [1537866370.065684758]: Messages of type 1 arrived out of order (will print only once)

```
and in rviz, pointcloud is noisy as below.
[image]

I have searched many similar issues and tried measurement to solve the problem.
But have not solved yet.

Could you tell me how to solve this problem?

SteveMacenski, 2018-09-25:
That looks like the typical default configuration.

dorodnic, 2018-09-27:
The latest release adds post-processing that can help with the noise. 
The noise is proportional to distance square, and is reduced when running dynamic calibration App.
The more the two sensors are out of calibration, the more Z noise you should expect to get.

hi-kondo, 2018-10-01:
@dorodnic 
Thank you for advice.
I'll update the latest release and try.

MaEppl, 2018-10-28:
But right now post-processing is just working for the rs_camera, not for the aligned_depth?

hi-kondo, 2018-11-09:
I use this launch command,
`roslaunch realsense2_camera rs_rgbd.launch align_depth:=true`
and set some parameters.
I show json_file.
```
{
    "aux-param-autoexposure-setpoint": "400",
    "aux-param-colorcorrection1": "0.461914",
    "aux-param-colorcorrection10": "-0.553711",
    "aux-param-colorcorrection11": "-0.553711",
    "aux-param-colorcorrection12": "0.0458984",
    "aux-param-colorcorrection2": "0.540039",
    "aux-param-colorcorrection3": "0.540039",
    "aux-param-colorcorrection4": "0.208008",
    "aux-param-colorcorrection5": "-0.332031",
    "aux-param-colorcorrection6": "-0.212891",
    "aux-param-colorcorrection7": "-0.212891",
    "aux-param-colorcorrection8": "0.68457",
    "aux-param-colorcorrection9": "0.930664",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "False",
    "controls-autoexposure-manual": "45715",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "166",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-depth-white-balance-auto": "False",
    "controls-laserpower": "300",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-autoexposure-setpoint": "400",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "24",
    "param-maxscorethreshb": "2047",
    "param-medianthreshold": "500",
    "param-minscorethresha": "1",
    "param-neighborthresh": "7",
    "param-raumine": "1",
    "param-rauminn": "1",
    "param-rauminnssum": "3",
    "param-raumins": "1",
    "param-rauminw": "1",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0499022",
    "param-regioncolorthresholdg": "0.0499022",
    "param-regioncolorthresholdr": "0.0499022",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "10",
    "param-robbinsmonroincrement": "10",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "325",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-width": "1280"
}
```

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@hi-kondo Did you try dynamic calibration for the camera to try again? Please refer to https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API to get the dynamic calibration tool and user guide. Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@hi-kondo If no other questions, will close the ticket accordingly.

closed 2019-05-03 by RealSenseCustomerSupport