realsenseai/realsense-ros #484: error: ‘invoke’ was not declared in this scope

5yler, 2018-09-28:
Trying to build the latest on `developent` branch (`31a0e4db8f2e9ad0349a37d97a8e7832591112d2`) on Ubuntu 16.04 with ROS kinetic. I previously built successfully with an older commit `ef2ee35`. `catkin_make` gives the following output:

```
[ 99%] Building CXX object realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o
[100%] Building CXX object realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/rs435_node.cpp.o
[100%] Building CXX object realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o
[100%] Building CXX object realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/sr300_node.cpp.o
[100%] Building CXX object realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/rs415_node.cpp.o
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:382:0: warning: ignoring #pragma omp parallel [-Wunknown-pragmas]
 #pragma omp parallel for schedule(dynamic)
 ^
In file included from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/rs415_node.h:6:0,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/rs415_node.cpp:1:
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h: In member function ‘void realsense2_camera::PipelineSyncer::operator()(rs2::frame) const’:
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:59:23: error: ‘invoke’ was not declared in this scope
    invoke(std::move(f));
                       ^
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:59:23: note: suggested alternative:
In file included from /usr/include/boost/thread/detail/thread.hpp:23:0,
                 from /usr/include/boost/thread/thread_only.hpp:22,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /usr/include/boost/thread.hpp:13,
                 from /opt/ros/kinetic/include/diagnostic_updater/diagnostic_updater.h:48,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:12,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/rs415_node.h:6,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/rs415_node.cpp:1:
/usr/include/boost/thread/detail/invoke.hpp:131:17: note:   ‘boost::detail::invoke’
     inline auto invoke(BOOST_THREAD_RV_REF(Fp) f, BOOST_THREAD_RV_REF(Args) ...args)
                 ^
In file included from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:1:0:
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/base_realsense_node.h: In member function ‘void realsense2_camera::PipelineSyncer::operator()(rs2::frame) const’:
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/base_realsense_node.h:59:23: error: ‘invoke’ was not declared in this scope
    invoke(std::move(f));
                       ^
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/base_realsense_node.h:59:23: note: suggested alternative:
In file included from /usr/include/boost/thread/detail/thread.hpp:23:0,
                 from /usr/include/boost/thread/thread_only.hpp:22,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /usr/include/boost/thread.hpp:13,
                 from /opt/ros/kinetic/include/diagnostic_updater/diagnostic_updater.h:48,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/base_realsense_node.h:12,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/boost/thread/detail/invoke.hpp:131:17: note:   ‘boost::detail::invoke’
     inline auto invoke(BOOST_THREAD_RV_REF(Fp) f, BOOST_THREAD_RV_REF(Args) ...args)
                 ^
In file included from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/rs435_node.h:6:0,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/rs435_node.cpp:1:
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h: In member function ‘void realsense2_camera::PipelineSyncer::operator()(rs2::frame) const’:
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:59:23: error: ‘invoke’ was not declared in this scope
    invoke(std::move(f));
                       ^
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:59:23: note: suggested alternative:
In file included from /usr/include/boost/thread/detail/thread.hpp:23:0,
                 from /usr/include/boost/thread/thread_only.hpp:22,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /usr/include/boost/thread.hpp:13,
                 from /opt/ros/kinetic/include/diagnostic_updater/diagnostic_updater.h:48,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:12,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/rs435_node.h:6,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/rs435_node.cpp:1:
/usr/include/boost/thread/detail/invoke.hpp:131:17: note:   ‘boost::detail::invoke’
     inline auto invoke(BOOST_THREAD_RV_REF(Fp) f, BOOST_THREAD_RV_REF(Args) ...args)
                 ^
In file included from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/sr300_node.h:6:0,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/sr300_node.cpp:1:
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h: In member function ‘void realsense2_camera::PipelineSyncer::operator()(rs2::frame) const’:
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:59:23: error: ‘invoke’ was not declared in this scope
    invoke(std::move(f));
                       ^
/home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:59:23: note: suggested alternative:
[... 21 lines cut ...]
                 from /usr/include/boost/thread.hpp:13,
                 from /opt/ros/kinetic/include/diagnostic_updater/diagnostic_updater.h:48,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:12,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/../include/sr300_node.h:6,
                 from /home/alfred/catkin_ws/src/realsense/realsense2_camera/src/realsense_node_factory.cpp:5:
/usr/include/boost/thread/detail/invoke.hpp:131:17: note:   ‘boost::detail::invoke’
     inline auto invoke(BOOST_THREAD_RV_REF(Fp) f, BOOST_THREAD_RV_REF(Args) ...args)
                 ^
realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:88: recipe for target 'realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/rs415_node.cpp.o' failed
make[2]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/rs415_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:101: recipe for target 'realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/rs435_node.cpp.o' failed
make[2]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/rs435_node.cpp.o] Error 1
realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:114: recipe for target 'realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/sr300_node.cpp.o' failed
make[2]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/sr300_node.cpp.o] Error 1
realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:62: recipe for target 'realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o' failed
make[2]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o] Error 1
realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:75: recipe for target 'realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
CMakeFiles/Makefile2:9978: recipe for target 'realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j32 -l32" failed
```

hi-kondo, 2018-09-29:
I also got the same error "error: ‘invoke’ was not declared in this scope" on `development` branch and Intel® RealSense™ SDK 2.0 (build 2.15.0) .
So updated latest SDK(build 2.16.0), `catkin_make` is succeeded.

5yler, 2018-09-29:
Thanks! That solved the issue.

closed 2018-09-29 by 5yler