realsenseai/realsense-ros #485: D435 wrong time stamp

lyziew, 2018-09-29:
Version |  Configuration
-- | --
Operating System | Ubuntu 16.04 LTS
Kernel | 4.4.0-135-generic
ROS | kinetic
ROS RealSense | 2.0.4
librealsense2 | 2,16.0
when i use rosbag record the /camera/color/image_rect_color toptic
i found the image header stamp is wrong 
when i use rqt_bag to see the bag , is normal
[image]
but when i subscribe it 
you can see
[image]

doronhi, 2019-01-28:
Hi @lyziew ,
Sorry for the long long time. 
If the problem still is relevant, could you please describe the exact commands you use for running realsense2_wrapper and the the rosbag in order to produce the printouts you published?
I'm having problems reproducing the problem.

scottnothing, 2019-02-11:
1) Does this only happen after running the camera for a long time?
2) What is `enable_sync` parameter value?

SteveMacenski, 2019-02-11:
I also see this issue on occasion

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

lyziew, 2019-03-11:
Sorry for now relpy to your message
As the computer system has been reset, so i can not repeat the bug
I remember I install sdk use apt command and use "rs_rgbd.launch" to start realsense
the bug only appear on my laptop before i reinstall system
and the renalsense work well after reinstall system
really really thank you very much

SteveMacenski, 2019-03-11:
I ended up having to edit the driver and remove the time-syncing attempts from the code. It only syncs once on bring up so its going to naturally drift alot over time. It needs to be either actively managed or removed all together

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

Is the issue still reproduced in the latest ros wrapper and librealsense?

tispratik, 2020-03-06:
This issue is still happening on 5.12.02.100 as well. The camera color/depth frame timestamps are way off than the the system time from host os. The difference is sometimes positive, sometimes negative. The frame timestamps from the camera are all over the place.

doronhi, 2020-03-09:
Hi, @tispratik 
To make it possible to investigate, could you please elaborate? What system you run on, versions, command line, specific example, any suspicious messages etc.?
Thanks.

tispratik, 2020-03-11:
Hello @doronhi, we installed realsense using sources and enabled the meta data flag. Prior to installing using sources and metadata flag, we the camera was giving random timestamps. After the command line install with metadata flag and kernel patches for Ubuntu 16.04, we started seeing less issues. But we consistently see anywhere from 30 to 80 milliseconds of difference between the camera timestamp and the host timestamp.

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.03.00| 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-88-generic                                         | 
| Platform                                 |    - |
| SDK Version                            |  -                          | 
| Language                            |  python                         | 
| Segment			|  Robot                          |

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

@tispratik Could you please tell the SDK version and ROS wrapper version when you get the issue? Thanks!

tispratik, 2020-03-23:
We are not using ROS and we simply compiled librealsense from stable source.

RealSenseCustomerSupport, 2020-03-24:
----------------------------------------------

@tispratik  Could you please tell the SDK version when get the issue? And could you please also provide the timestamp logs for the issue? Thanks!

tispratik, 2020-03-24:
rs.__version__ gives 2.33.1. 
The wrong timestamps went away when we compiled from source.

RealSenseCustomerSupport, 2020-03-25:
----------------------------------------------

@tispratik Does it mean that you didn't get the issue again with the latest v2.33.1 SDK? Looking for your clarification. Thanks!

tispratik, 2020-03-25:
That is correct.

RealSenseCustomerSupport, 2020-03-26:
----------------------------------------------

@tispratik Glad to hear the issue resolved. Will close the issue accordingly. Thanks!

closed 2020-03-26 by RealSenseCustomerSupport