realsenseai/realsense-ros #486: D435 Camera Serial Number

rupak-d, 2018-10-01:
Hi, 
Is it possible to automate getting the camera serial numbers for the multiple camera launch file?
Thanks!

doronhi, 2018-10-09:
I also feel it's a hassle to write the serial number manually every time but, assuming there are 2 cameras connected, how will you decide automatically which camera will have a group namespace of camera1 and which camera2?
And if you have more than 2 cameras attached?

rupak-d, 2018-10-09:
I am using 3 cameras in different orientations. The plan is to use this to decide the namespaces.
Or if connecting the cameras in a specific order on the USB hub matches the ordering with the output of
string serial = dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
that might be the easiest solution.

mikepurvis, 2019-03-01:
I'd like to extract the serial number based on which port each one is connected to, which should be possible by just reading it out of the filesystem, eg:

```
$ cat /sys/bus/usb/devices/usb4/4-4/serial 
830513024396
```

However, for reasons which are not clear to me, the serial numbers returned by the USB subsystem don't match the ones which come back from `dev.get_info`. I'm assuming this is simply a bug in the current firmware.

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

ihadzic, 2019-11-20:
I also needed this because I find it easier to tell the devices apart by the USB path than the serial number (in the former case I only have to make sure that wiring is done in predictable manner, but I can swap the cameras in and out without changing the configuration. I came up with this one-liner that helps me determine the serial number to pass to roslaunch:
```
rs-enumerate-devices | grep 'Serial Number' | paste - - | grep $(cat /sys/bus/usb/devices/$USB_PATH/serial) | awk '{print $4}'
```

where `$USB_PATH` is something like `3-2` (e.g., bus 3, port 2). The `serial` file under `sys` is apparently ASIC serial number instead of the device serial number and hence the round-about way above.

If someone has a more simpler way of doing this, I would appreciate the enlightenment.

mikepurvis, 2019-11-21:
@ihadzic An even simpler version of this is to just use roslaunch's built-in ability to assign the output of a command invocation to a parameter. However, both approaches have the issue that you get a race on startup where if the camera doesn't enumerate in time, the driver will never find it (or worse, the whole launch fails).

Anyway, all of this is irrelevant now that the driver natively supports `usb_port_id`, so you should switch to that, and this ticket can be closed. FYI @RealSenseCustomerSupport

closed 2020-01-08 by RealSenseCustomerSupport

ShubhamKPBoss, 2024-01-04:
Hello, can you please help me how to connect ip camera using camera --> serial number ,port , username, password
which type of follow protocol ............... ?

monajalal, 2025-02-28:
@MartyG-RealSense 

rs-enumerate-devices | grep 'Serial Number' | awk -F': ' '{print $2}'
I get two serial number.

(my) root@DOS:/home/mona/FoundationPose# rs-enumerate-devices | grep 'Serial Number'                                                        
    Serial Number                 : 	250122070375
    Asic Serial Number            : 	415743023395

MartyG-RealSense, 2025-02-28:
@monajalal  If you want the camera serial number then 'Serial Number' is the one to use.

The Asic refers to the 'Vision Processor D4' circuit board inside the camera.

monajalal, 2025-02-28:
@MartyG-RealSense Thanks a lot for your clarification.