realsenseai/realsense-ros #488: List of dynamic_reconfigure parameters is brittle

AndyZe, 2018-10-02:
When I tried to add a new dynamic_reconfigure parameter for PR487, I eventually figured out that the "level" of the parameter in _base_d400_params.py_ needs to match the order of the enum in base_realsense_node.h. This seems confusing and brittle.

For example, this dynamic parameter won't work as expected:
`  gen.add(str(prefix) + "my_parameter",                          str_t,    10,  "my_parameter",            "")`

If the enum is declared in this order:
`    enum base_depth_param{
        base_depth_gain = 1,
        **base_my_parameter**,   // <-- Note, it's declared second
        base_depth_enable_auto_exposure,
        base_depth_visual_preset,
        ...`

doronhi, 2019-01-07:
@AndyZe , did you check out the latest release?

AndyZe, 2019-01-07:
Was this fixed by switching to ddynamic_reconfigure? Anyway - nice!

closed 2019-01-07 by AndyZe