realsenseai/realsense-ros #491: Depth height and width zero if enable pointlcoud set to true

atyshka, 2018-10-11:
I'm running the latest version of this ros package with librealsense 1.16 and I'm running into issues when trying to use the pointcloud. If I launch rs_rgbd normally, I get a correct depth stream I can visualize in RVIZ. However, if I use the same launch file with the `enable_pointcloud` arg set to true, suddenly the width, height, and step of the depth image are zero. The actual image array is fully published, but without a height, width, and step this data is useless. Also, rs_aligned works fine without the pointcloud enabled, but with it enabled I get ` An error has occurred during frame callback: null pointer passed for argument "frame_ref"` when echoing the topic. Is this a known issue? A few versions back I could use both pointcloud and depth just fine.

Output of `rostopic echo /camera/depth/image_raw --noarr` with cloud disabled:
`header: 
  seq: 5792
  stamp: 
    secs: 1539267647
    nsecs: 369592057
  frame_id: "camera_depth_optical_frame"
height: 480
width: 640
encoding: "16UC1"
is_bigendian: 0
step: 1280
data: "<array type: uint8, length: 614400>"`

and with cloud enabled:

`header: 
  seq: 95
  stamp: 
    secs: 1539268016
    nsecs: 367752813
  frame_id: "camera_depth_optical_frame"
height: 0
width: 0
encoding: "16UC1"
is_bigendian: 0
step: 0
data: "<array type: uint8, length: 614400>"`

arb93, 2018-10-16:
Same here. Any suggestions?

atyshka, 2018-10-16:
Related to #489

atyshka, 2018-10-18:
@doronhi Update: As of your latest commit, height and width work fine on the depth image aligned topics, but with enable_pointcloud set to true  `/camera/depth/image_raw` still has height and width zero.

doronhi, 2018-10-21:
Last commit fixes this issue.

closed 2018-10-23 by atyshka

leiw324, 2019-02-23:
I am getting this issue too, how to fix ?

doronhi, 2019-02-24:
What versions do you use? librealsense and realsense2_camera?