realsenseai/realsense-ros #492: WARNING [139954094954240] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details

weiwuhuhu, 2018-10-11:
hello,
     when i execute the roslauch realsense2_camera rs_rgbd.launch ,the warning has occurred,

             11/10 22:03:53,863 WARNING [139954094954240] (ds5-timestamp.cpp:64) UVC metadata                                    payloads not available. Please refer to installation chapter for details.
             [ WARN] [1539266633.868420728]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME).
            What's influence of this warning.

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

closed 2019-04-11 by RealSenseCustomerSupport

jarvis-owl, 2019-09-20:
Hej @SuperKuooo ,
if you still care:  the topics are being published though.
You can watch the stream with `rqt_image_view`

BriceRenaudeau, 2020-01-13:
Hello, I hade the same problem :
WARNING [...] (ds5-timestamp.cpp:64) UVC metadata payloads not available
[ WARN] [/camera_right/realsense2_camera_manager]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)

[Ubuntu 18.04, kernel 5.0, melodic, D435, firmware 5.12, librealsense2-dkms up-to-date]

I hade no issues until today where I found that the TimeStamp in the depth images published by my camera where stuck in the past.
The stamps were oscillating in the past (not fixed).
Strangely when I added `<arg name="filters" default="pointcloud"/>` in the launch file, the stamps were good again.

doronhi, 2020-01-30:
The timestamps issue in kernel 5 could be related to #1047.

szx0112, 2020-02-26:
I got the similar warning in realsense-viewer with librealsense 2.32.1. I installed the dkms, utils and dev packages. My pc is intel nuc 8 and ubuntu 18.04, kernel 5.3.0-40-generic

Any suggestions?