realsenseai/realsense-ros #493: why pointcloud data set to (0,0,0) if z>5.0f?

ahuizxc, 2018-10-16:
cannot understand why set this threshold, it can cause bad influence while displaying the pointcloud data in rviz.

doronhi, 2018-10-25:
We no longer use the specified piece of code if I close it as irrelevant.

closed 2018-10-25 by doronhi

ahuizxc, 2018-10-26:
We need this topic that subscribed by rviz, and the pointcloud displayed in rviz is very terrible!

doronhi, 2018-10-26:
What version of realsense2_camera are you using? The latest version use the pointcloud function from librealsense so if you can. A first good step would be to update both.
As for bad looking pointcloud, it means poor depth performance and could be caused by several factors. I would start by checking out this paper:
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

freemanlo, 2018-10-26:
Hi @ahuizxc,

1. What's your required use range? 
2. You can consider to change the preset (ex: high density for depth) to see how it works.
3. You can also increase the laser projector power if a long distance is required.