realsenseai/realsense-ros #494: Split description from ROS driver

wxmerkt, 2018-10-16:
Hi,
We have many use cases where we'd just like to have the description (xacro/URDF/mesh) on a machine and not have the requirement to install the SDK and compile the driver. Thus it would be great to create (and binary-release) a ``realsense_description`` package which includes the xacro, URDF, visualisation-launch file, and the meshes. Plus, the current DAE is 15 MB (!) - a watertight and shrunk smaller mesh would be better.

On our end, we'd be happy to do the work towards creating a ``realsense_description`` package and binary-release it if there is interest from the community.

Thanks,
Wolfgang

AustinDeric, 2018-11-16:
In addition, i have a few observations/requests 
-  there is not ${prefix} in the xacro, which makes running 2 cameras in the same urdf impossible (http://wiki.ros.org/urdf/Tutorials/Using%20Xacro%20to%20Clean%20Up%20a%20URDF%20File)
- The xacros are not available to the users of the binary because the urdf directory is not installed: https://github.com/plusone-robotics/realsense/blob/development/realsense2_camera/CMakeLists.txt#L120.  Was there a reason for this.

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

athackst, 2019-03-12:
I'm also interested in this.  We've internally forked this and made changes for our system (including adding gazebo tags).  @AustinDeric @wxmerkt  I'm happy to contribute if you've already started something

wxmerkt, 2019-03-12:
I'd like to see some input and feedback from the @intel-ros maintainers of this driver, e.g., @doronhi - we seem to have strong community interest in improvements to the description and splitting it out/releasing it to the ROS buildfarm. Can you please comment/assist? Thank you!

jarvisschultz, 2019-03-20:
I also just finished adding Gazebo tags and prefix support to the xacro files for an internal project here. Seems like there is interest in making these changes. Happy to support any PRs if they'd be accepted.

athackst, 2019-03-20:
Kk, I made a minimal change set PR so we can separate the reorg from the URDF updates.

closed 2019-07-14 by doronhi