realsenseai/realsense-ros #496: Reddish images

vossie11, 2018-10-19:
The images captured by the RGB camera seems to be reddish in ROS. I used the rs_rgbd.launch file.

[image]

When i use RealSense-viewer to capture a image (Windows or Ubuntu) the image appears to be normal. 
[image]

I did find a temporarily solution. The issue is fixed by commenting out the rs435_color_white_balance case in rs435_node.cpp

Is this a known problem?

RealSense D435
ROS Kinetic
Ubuntu 16.04
Librealsense v2.16.1

doronhi, 2018-10-21:
I don't exactly see the issue. I run ros and watch the image with rqt_image_view and then run the realsense_viewer and see, what seems to me, the same image.
Instead of commenting out the white balance section in rs435_node.cpp, couldn't you adjust the values using 
`rosrun rqt_reconfigure rqt_reconfigure 
`
to get the image you want?

vossie11, 2018-10-22:
Adjusting the values doesn't seem to help. 
Can you try the following steps to reproduce the issue?
1. Disconnect and connect the realsense (make sure old parameters are cleared)
2. Run rs_rgbd.launch with rqt_image_view
3. Disconnect and connect the realsense (clear parameters set by ros)
4. Run realsense_viewer

The connect and disconnect steps seems to be important to reproduce the issue.

atyshka, 2018-10-23:
I can confirm that my realsense by default also has a reddish tint but dragging the white balance int rqt fixes it for me

doronhi, 2018-10-25:
[image]

These are the 2 images I acquired. I don't see the redish thing. I did managed to see a noticeable difference if, in the realsense-viewer, I turned off the "Enable auto white balance" flag and made the realsense-viewer's image redder (? more redish...?)
(That, however, revealed a bug that the flag is not available for D435 in the realsense2_camera ros-wrapper. Will be fixed shortly).
However, since the default value for "Enable auto white balance" is "On" (comes from the firmware), unless you turn it off with a tool other then the ros-wrapper, (like I did with realsense-viewer), I can't see how to reproduce the problem that the image, coming out of the ros-wrapper, is redish compared to the one seen in realsense-viewer.

iamtesch, 2018-12-04:
I am experiencing the same behavior -- reddish tint in ROS, not in realsense-viewer.

RealSense D435
ROS Melodic
Ubuntu 18.04
Librealsense v2.17
ROS librealsense wrapper v2.1.1 (with patch applied at https://github.com/intel-ros/realsense/pull/523/files to allow 2.1.1 to compile with librealsense 2.17)

closed 2019-01-27 by doronhi