realsenseai/realsense-ros #501: Filters not working?

atyshka, 2018-10-23:
I'm a little confused about the new filters feature? What topics exactly are they supposed to affect in the and how? The readme is rather vague about this. For example, I would expect the colorizer to ouput a colored depth image kind of like with the realsense viewer. However, I don't see any effect from this filter or any of the others.

doronhi, 2018-10-25:
Thank you for the comment. I updated the readme a little, I hope its a little clearer now.
The colorizer filter replace the depth image topic. In this case the image will be RGB instead of the normal uint_16.
The temporal and spatial filters affect the depth image and it's derivatives. They do not produce new topics of their own. Currently their inner parameters are fixed to their default values and the effect is not easy to visually detect. I am currently working on a version that will allow modifying them. In the meanwhile you can check the effect of playing with them, using the realsense-viewer application.

atyshka, 2018-10-25:
Colorizer not working here:
[image]
after launching with `roslaunch realsense2_camera rs_aligned_depth.launch enable_pointcloud:=true filters:=colorizer`

atyshka, 2018-10-25:
Oh, I see. You only updated the rs_camera launch file

ManMan88, 2019-04-02:
I've tried using the filters, and I couldn't see their effect (more specifically - the spatial effect)

@doronhi 
> The temporal and spatial filters affect the depth image and it's derivatives. They do not produce new topics of their own.

When using the spatial filter, I assume the depth image resolution should decrease compared to the original one. However, when I check what is the size of the image published in `/camera/depth/image_rect_raw`, it is the same size as the original image.
When I check the spatial filter with the realsense-viewer, I can see that the image resolution decreases.

So I couldn't fing how to access the filtered data.
Or am I missing something?

doronhi, 2019-04-03:
I can't see the problem. When I play around with filter_smooth_delta I see it's effect.
The image stays with the same resolution despite the filter. Both in the viewer and in realsense2_camera.

ManMan88, 2019-04-03:
Sorry, I was wrong, I meant the Decimation filter.
I can clearly see that when using it in viewer that the resolution decreases.
For example with a factor of 2, the resolution becomes 640X360 (if it was 1280X720 originally)

ManMan88, 2019-04-03:
O.K, I found the problem.
The decimation filter does work. 
However, when it is combined with the disparity filter, the resolution stays high - so it seams that the the decimation does not work with it (maybe other filters as well, I didn't check)

abhijitmajumdar, 2019-04-18:
@ManMan88 I am facing the exact same issue. `disparity` works only when not using the `decimation` filter. However, when I run the `rs-post-processing` example, it seems that the disparity does not affect the decimation. Further looking into the [details of the suggested implementation](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#using-filters-in-application-code) indicated that the `decimation` should be applied before the `disparity` (start and end). However, if you notice the `disparity` start occurs before the `decimation` in this ros wrapper, which can be seen in the code [here](https://github.com/intel-ros/realsense/blob/development/realsense2_camera/src/base_realsense_node.cpp#L918) as well as the the [logs you posted](https://github.com/intel-ros/realsense/issues/501#issuecomment-479014599)

@doronhi any update on this?

abhijitmajumdar, 2019-04-20:
@doronhi I have a potential fix for this issue [here](https://github.com/intel-ros/realsense/pull/737)

doronhi, 2019-05-22:
> @doronhi I have a potential fix for this issue [here](https://github.com/intel-ros/realsense/pull/737)

Thank you. Merged.

closed 2019-05-22 by doronhi

ranjitkathiriya, 2020-11-20:
Hi @doronhi,

Actually, I am getting RGB-D from the stream but it is in black color can you please help me with that here I am posting my code and Image of the output.

```
#!/usr/bin/env python3
import rospy
from sensor_msgs.msg import Image
from cv_bridge import CvBridge 
import cv2
import numpy as np
 
def callback(data):
  br = CvBridge()
  rospy.loginfo("receiving video frame")
  current_frame = br.imgmsg_to_cv2(data)
  cv_image_array = np.array(current_frame, dtype = np.dtype('f8'))
  cv_image_norm = cv2.normalize(cv_image_array, cv_image_array, 0, 1, cv2.NORM_MINMAX)
  cv2.imshow("camera", cv_image_norm)
  print(cv_image_norm.shape)
  cv2.waitKey(1)
      
def receive_message():
  rospy.init_node('video_sub_py', anonymous=True)
  rospy.Subscriber('/camera/aligned_depth_to_color/image_raw', Image, callback)
  rospy.spin()
  cv2.destroyAllWindows()
  
if __name__ == '__main__':
  receive_message()
```

[image]

Thank you so much for the help

Actually, I have checked the launch file I don't understand what I have to change to get RGB D ?

ranjitkathiriya, 2020-11-20:
Thanks, Solved::

by running this ros command:
roslaunch realsense2_camera rs_camera.launch enable_pointcloud:=true filters:=colorizer align_depth:=true

shrutic12, 2023-09-21:
Hi! I am new to the fitlers and I'm wondering how can we set the parameters for spatial and temporal filtering from command line? 
For instance if I want to have something like:
roslaunch realsense2_camera rs_camera.launch enable_pointcloud:=true filters:=spatial,temporal align_depth:=true

How can I also put the values of the filter parameters?

MartyG-RealSense, 2023-09-21:
Hi @shrutic12  In ROS1 you can set the general filter type in the roslaunch instruction, and then set the specifics of the filter in a **rosrun** instruction in the terminal after the roslaunch command, using dynamic reconfigure.
  For example for the Decimation filter:

`rosrun dynamic_reconfigure dynparam set camera/decimation 'Filter Magnitude' 4`

Please see https://github.com/IntelRealSense/realsense-ros/issues/583 for more information.